loicland/superpoint_graph #107: error in running learning/custom_dataset.py to preprocess the dataset

amardeepjaiman, 2019-04-03:
Hi @loicland ,

I have partitioned my custom dataset using partition.py. I am having around 21 clouds in training set and 8 clouds in test set. After this i tried to run learning/custom_dataset.py as suggested. But I am getting following error:

000018.h5
(540866, 3)
(542251, 3)
(540866, 5)
Traceback (most recent call last):
  File "learning/custom_dataset.py", line 115, in <module>
    preprocess_pointclouds(args.CUSTOM_SET_PATH)
  File "learning/custom_dataset.py", line 96, in preprocess_pointclouds
    P = np.concatenate([xyz, rgb, elpsv], axis=1)
ValueError: all the input array dimensions except for the concatenation axis must match exactly

Could you please let me know, what kind of changes are required in this file. I also have RGB in my data so thats not a problem.

amardeepjaiman, 2019-04-03:
Hi @loicland 

I also checked my dataset after pruning. The feature/*.h5 and found out that i have lost the lables value after pruning. I debug the code and found out that lables got read successfully on below line :
xyz, rgb, labels = read_ply(data_file)

but after libply_c.prune() function , all the lable values changed to 0.

xyz, rgb, labels = libply_c.prune(xyz, args.voxel_width, rgb, np.array(1,dtype='u1'), 0)

could you suggest what could be the issue ?

loicland, 2019-04-04:
Hi,

it seems that somehow your rgb data is of different size than the xyz. It shouldn't happen.

can you check that after: `xyz, rgb, labels = read_ply(data_file)` the arrays are consistent in size? What about just after the pruning?

As for the labels being zero,

`xyz, rgb, labels = libply_c.prune(xyz, args.voxel_width, rgb, np.array(1,dtype='u1'), 0)` wil produce 0 labels. 

if you want to compute the labels histograms per voxel, you need to use:

`xyz, rgb, labels = libply_c.prune(xyz, args.voxel_width, rgb, labels, n_labels)`

closed 2019-12-23 by loicland