loicland/superpoint_graph #111: pytorch problem

lapetite123, 2019-04-10:
Hi!
I followed the official instruction to install pytorch using 'conda install pytorch torchvision cudatoolkit=10.0 -c pytorch' (my cuda version is 10.0)
but I find my torch version is '1.0.1.post2'. I am wondering that while it's right or not?
I also tried to install pytorch followed your instruction, the torch version is '1.0.1.post2' ,too.
I don't know how t solve it, can you please help me. Thanks.

loicland, 2019-04-10:
SPG should be compatible with 1.0. The only problem will be with in `./learning/modules.py`.

Apply the patch proposed in this issue https://github.com/loicland/superpoint_graph/issues/98

Let me know if it works for you!

closed 2019-04-23 by loicland

lapetite123, 2019-04-27:
Hi，I‘m so sorry for too late to reply. Besides #98 ,thanks to version difference, I also changed line 203 in main.py as:
logging.debug('Batch loss %f, Loader time %f ms, Trainer time %f ms.', loss.item(), t_loader, t_trainer) 

Another question. during training, it stopped because of out of memory(gpu is 10GB), I changed line 132-133 in pointnet.py as to solve the problem:
        with torch.no_grad():
            out = model.ptn(Variable(clouds), Variable(clouds_global))
        with torch.no_grad():
            out = Variable(out.data, requires_grad=model.training)

another problem during training:
100%|█████████████████████████████████████████████████████████████████| 4/4 [00:38<00:00,  8.61s/it]
  0%|                                                                         | 0/4 [00:00<?, ?it/s]/home/anaconda3/envs/spg/lib/python3.6/site-packages/torch/nn/functional.py:1332: UserWarning: nn.functional.sigmoid is deprecated. Use torch.sigmoid instead.
  warnings.warn("nn.functional.sigmoid is deprecated. Use torch.sigmoid instead.")
/home/anaconda3/envs/spg/lib/python3.6/site-packages/torch/nn/functional.py:1320: UserWarning: nn.functional.tanh is deprecated. Use torch.tanh instead.
  warnings.warn("nn.functional.tanh is deprecated. Use torch.tanh instead.")
100%|█████████████████████████████████████████████████████████████████| 4/4 [00:40<00:00,  9.05s/it]
/home/superpoint_graph/learning/metrics.py:66: RuntimeWarning: invalid value encountered in true_divide
  return sum(values) / class_seen

I don't have any idea about the error in metrics.py

jeyden, 2019-06-23:
Hello, I encountered same problem about metrics.py on pytorch=1.0.1.  Have u solved it？Thx.

bimtauer, 2019-07-10:
I found 'sg28_Station2' from the test_reduced set responsible for the memory spike - I assume this is because it contains it much larger number of clusters (25724 compared to the 5k and below for the other files from test_reduced. I have removed the respective files for now but will try  lapetite123's torch.no_grad() fix next. 

As for the metric issue: I also get the following warning in the first evaluation round:
RuntimeWarning: invalid value encountered in true_divide
  return sum(values) / class_seen

But isn't this due to the fact that (at least when training with the arguments from the readme, will say with _--db_test_name testred --db_train_name trainval_) we train on _training_ + _validation_ but test on the unlabelled _test_reduce_? 

When training with _--db_test_name val --db_train_name train_ we're using an 11/4 split of the labeled train data as specified in sema3d_dataset.py lines 22-23 - then there is no divide error