loicland/superpoint_graph #112: Questions in ply_c.cpp about std::numeric_limits<float>::min()

Yajha, 2019-04-15:
Hi,

when i search numeric_limits in [http://www.cplusplus.com/reference/limits/numeric_limits/](url)
it says lowest() means the negative of max(),so i wonder why did you use min() in the voxelization process in ply.cpp ?

Thanks for your time.

loicland, 2019-04-29:
Hi,

when comparing finite values, `min` and `lowest` have the same behavior.

Yajha, 2019-05-03:
Hi, loic.

But i found some scenes which has negative coordinates 
(for example, in S3DIS dataset Area 1 office_31)

Also, i have compared the different results and there are just a little difference. I suspect that it is because 
there are just a few negative coordinates

Thanks for your time.

loicland, 2019-05-03:
I think you're right. Thank for pointing this out, I will fix it ASAP.

closed 2019-05-09 by Yajha