loicland/superpoint_graph #114: Different results on S3dis dataset

micaeltchapmi, 2019-04-17:
Hi, 

Why do I get different results each time I train the network on the s3dis dataset. Initially, I downloaded the code, followed the instructions and ran:

for FOLD in 1 2 3 4 5 6; do \
CUDA_VISIBLE_DEVICES=0 python learning/main.py --dataset s3dis --S3DIS_PATH $S3DIR_DIR --cvfold $FOLD --epochs 350 --lr_steps '[275,320]' \
--test_nth_epoch 50 --model_config 'gru_10_0,f_13' --ptn_nfeat_stn 14 --nworkers 2 --odir "results/s3dis/best/cv${FOLD}"; \
done

I then ran this for testing:

for FOLD in 1 2 3 4 5 6; do \
CUDA_VISIBLE_DEVICES=0 python learning/main.py --dataset s3dis --S3DIS_PATH $S3DIR_DIR --cvfold $FOLD --epochs -1 --lr_steps '[275,320]' \
--test_nth_epoch 50 --model_config 'gru_10_0,f_13' --ptn_nfeat_stn 14 --nworkers 2 --odir "results/s3dis/best/cv${FOLD}" --resume RESUME; \
done

As I am interested in the Area 5 results, I report a mean iou of 58.1 which is close enough to the results in the paper. However, I trained the network again and test only on A5. So I run:
``
for FOLD in 5; do \
CUDA_VISIBLE_DEVICES=0 python learning/main.py --dataset s3dis --S3DIS_PATH $S3DIR_DIR --cvfold $FOLD --epochs 350 --lr_steps '[275,320]' \
--test_nth_epoch 50 --model_config 'gru_10_0,f_13' --ptn_nfeat_stn 14 --nworkers 2 --odir "results/s3dis/best/cv${FOLD}"; \
done
``

And then I run:

``
for FOLD in 5; do \
CUDA_VISIBLE_DEVICES=0 python learning/main.py --dataset s3dis --S3DIS_PATH $S3DIR_DIR --cvfold $FOLD --epochs -1 --lr_steps '[275,320]' \
--test_nth_epoch 50 --model_config 'gru_10_0,f_13' --ptn_nfeat_stn 14 --nworkers 2 --odir "results/s3dis/best/cv${FOLD}" --resume RESUME; \
done
``
After training and testing with the above two commands several times, I got A5 mean_ious of 54.52, 55.47,  56.3, 54.97 and other similar results which seem too far from the results in the paper. Can you help me understand why the discrepancy in the results? How do I make sure I get almost the same results at each run?

loicland, 2019-04-17:
Hi,

this is definitively an issue, throughout deep learning, but probably in particular for SPG. I think the pointNets are mostly stable, but the superedge filters are very unconstrained (in the m-v setting, generate 32*32 values from 13 superedge features) , and their influence are amplified at each iteration of ECC. Better superedge filters would probably help in this regards, it is in the list of things I need to explore.

The only workaround that I have found is to use a validation set within the training folds, and to take the epoch scoring best on the validation set. I will include this change in the next version as soon as I find some time. In the meantime I had success stabilizing the score on S3DIS with the following validation:

valid_names = ['hallway_1.h5', 'hallway_6.h5', 'hallway_11.h5', 'office_1.h5' \
                 , 'office_6.h5', 'office_11.h5', 'office_16.h5', 'office_21.h5', 'office_26.h5' \
                 , 'office_31.h5', 'office_36.h5'\
                 ,'WC_2.h5', 'storage_1.h5', 'storage_5.h5', 'conferenceRoom_2.h5', 'auditorium_1.h5']

taken in each folds of the training set. Let me know if it helps you as well.

micaeltchapmi, 2019-04-24:
Thanks for the reply. I changed the validation set to include only the valid_names you gave but I got an A5 mean_iou of 72.84% which is far beyond the A5 results in the paper. I ran the experiment a second time but I got 74.3% which shows that the results are still not consistent. 

I also tried including only the valid_names present in the Area 5 folder but I got an A5 mean iou of 52.2% and then 51.7% which are too low and also inconsistent. 

Can you share with me the results you obtained when you used valid_names as validation set? Is there any other way I can get the results in the paper and avoid inconsistent values?

mys007, 2019-04-24:
Hi,

thanks for the interesting comment. 

Regarding the variance of results, this comes from the stochastic training inherent to virtually all deep learning models. Some architectures are more susceptible to these effects than others. We have to admit that like many other papers, we have been guilty of not looking at the statistics of the performance - in fact, performance scores are distributions not numbers. A single number can be biased if the performance on validation and test set are highly correlated (I actually don't know how much is this the case for S3DIS, fold-wise). A more sound way of evaluating a novel architecture (with fixed hyperparameters) is to train multiple times and measure the confidence intervals. In this regard, I really like the nice line of argumentation in [A Boo(n) for Evaluating Architecture Performance](https://arxiv.org/abs/1807.01961), as well as the idea to report the expected maximum performance rather than the mean. Though of course the nicest would be to show a full histogram of performance scores. If there is time and resources, this is certainly a good thing to do in future papers.

Regarding your particular results, just to make sure that the worse performance is indeed due to training rather than some other issues, could you perhaps try to evaluate our pre-trained model available from [here](http://imagine.enpc.fr/~simonovm/largescale/models_s3dis.zip)?

Regarding the suggestion from Loic, I believe he meant using the "inner" validation set just as a stopping criterion. So once the training has finished, you can choose the model at the best epoch (not caring about the absolute performance there) and evaluate it on the test set.

micaeltchapmi, 2019-05-08:
Hi, 

Thanks for your reply and the good paper. First, I downloaded your model and evaluated it. I got an A5 mean iou of 57.5 instead of 58.04 as reported in the paper. Is there any particular reason why I do not get the exact reported mean iou?

Secondly, I tried what Loic suggested. I used the validation set he provided and then chose the model at the best epoch to evaluate on the test set. I am testing every 50 epochs and it's always the last epoch (epoch 349) that performs best so not surprising that I get similar results as before when only the model at the last epoch was evaluated. I ran the experiments three times and I got A5 mean ious of 56.7, 56.98, and 58.6. The results are still relatively unstable.

The idea presented in the paper will definitely be a more standard way of evaluting new models but that is not really useful to me at the moment. Have you thought of any other ways to make the results more stable? Please let me know!

Thanks!

drprojects, 2019-05-15:
Hi, 

I have been following this thread with interest, since I have been trying to use a randomly-sampled validation set for best-so-far model selection. But the stochasticity of my validation set sampling (10% of the training data picked) introduces additional variance in the testing performance and it is hard to track performance improvement.

Thanks a lot for recommending A Boo(n) for Evaluating Architecture Performance, it sure is an uplifting read !

closed 2019-06-26 by loicland