loicland/superpoint_graph #116: Auestion about  the label that is writen to fea file

yangLiu1022, 2019-04-19:
hi,
   when I read fea file,I find that the label is like the follow:
[[ 27636      0      0 ...      0      0      0]
 [277837      0      0 ...    155      9      0]
 [   135 932541      0 ...      0      0      0]
 ...
 [     0      0      0 ...      0      0      0]
 [     0      0     10 ...      0      0      0]
 [     0      0      0 ...      0      0      0]]
Is this one-hot label?
what's this represent?
I can't understand,could you help me understand,I'll be grateful！
thank you very much！

loicland, 2019-04-19:
Hi,

In `/partition/partition.py` a first step is the pruning of the point cloud
8n a regular voxel grid, controlled by the `--voxel_width` parameters.
Every subsequent step is applied to the pruned cloud.

However, in order to use the labeling of the original point cloud in the
evaluation metric, we keep track of the labels breakdown within each voxel.

The first column is the number of unannotated points (label 0) within
voxels. The other columns are the number of points in the other labels.

From the small extract that you posted  think that your `voxel_width` is
too large, you should have thousands of points within voxels. Use the
`visualize` method to see the pruned cloud.

Also does your graph has many unlabeled points? Reminder that the class 0
should be reserved for unlabeled data, while the prediction will shift the
labels by one (ie class 1 in the gut will be predicted as 0, see the
numerous issues on the subject).

yangLiu1022, 2019-04-19:
Thank you very much！

closed 2019-04-29 by loicland