loicland/superpoint_graph #117: Different result on semantic3d dataset

yangLiu1022, 2019-04-20:
Hi,
I run the code on semantic3d dataset,the result I get belows:
Multisample 10: Test accuracy: 86.70525808007719, 	Test oAcc: 0.9341802168018564, 	Test avgIoU: 0.7190518882340473, 	Test mAcc: 0.8343440083403026
I do not know the means of  "Test accuracy" and  "Test oAcc",and the difference between them.
And in your paper,dose the "OA" in the table represents "Test oAcc"?
Thank you very much.

loicland, 2019-04-26:
Hi,

`Test accuracy` is given by `tnt.meter.ClassErrorMeter`, I refer you to its documentation. It computes the overall precision.

`oAcc`, `avgIoU`, and `mAcc` are defined in `learning/metrics.py`. They denote respectively the overall accuracy (OA), the classwise mean IoU, and  the class wise mean accuracy (precision).

closed 2019-04-29 by loicland

yangLiu1022, 2019-04-30:
thank you very much，I will refer to the documentation.