loicland/superpoint_graph #119: None Object encountered when running vizualize.py on S3DIS

micaeltchapmi, 2019-04-22:
Hi,

After training the model on S3DIS, I get this error message when attempting to visualize the prediction file:

``
Traceback (most recent call last):
  File "partition/visualize.py", line 67, in <module>
    if (len(pred_red) != len(components)):
TypeError: len() of unsized object
``
I checked pred_red and it shows:                                                                                                                                                                                             
array(None, dtype=object)

I checked my results directory and the prediction file exists. Can you help me understand why I get this error and how I can successfully run partition/visualize.py?

loicland, 2019-04-26:
Hi,

can you post the exact command lie you used?

micaeltchapmi, 2019-05-08:
Hi

This is the exact command line:

`python partition/visualize.py --dataset s3dis --ROOT_PATH S3DIS --res_file 'results/s3dis/best/cv5/predictions_val' --file_path 'Area_1/conferenceRoom_1' --output_type igfpres`

and the Traceback:

`Traceback (most recent call last):
  File "partition/visualize.py", line 67, in <module>
    if (len(pred_red) != len(components)):
TypeError: len() of unsized object`

loicland, 2019-05-10:
Hi,

can you print the`pred_red.shape` and the `len(components)` just before the error please?

micaeltchapmi, 2019-05-11:
Hi,

This is what I get after printing pred_red and len(components) for the command above:

pred_red shape:  ()
components length:  786

loicland, 2019-05-16:
Ok the problem is that you are asking to output tmao room in area1 from the fold 5. Change either the `--res_file` or the `--file_path` so that the target room is indeed in the res_file.

micaeltchapmi, 2019-05-20:
You're right! It works now! Thanks!

closed 2019-05-20 by micaeltchapmi

longmalongma, 2019-12-10:
> Ok the problem is that you are asking to output tmao room in area1 from the fold 5. Change either the `--res_file` or the `--file_path` so that the target room is indeed in the res_file.

python partition/visualize.py --dataset sema3d --ROOT_PATH SEMA3D_DIR --res_f
ile 'results/sema3d/trainval_best/prediction_testred' --file_path 'test_reduced/MarketplaceFeldkirch_Station4' --output_type ifprs
Traceback (most recent call last):
  File "partition/visualize.py", line 70, in <module>
    raise ValueError("%s does not exist and is needed to output the result ply" % res_file) 
ValueError: results/sema3d/trainval_best/prediction_testred.h5 does not exist and is needed to output the result ply
can you help me?

longmalongma, 2020-04-22:
> Ok the problem is that you are asking to output tmao room in area1 from the fold 5. Change either the `--res_file` or the `--file_path` so that the target room is indeed in the res_file.

Hi,I also meet this problem,can you hep me?
  File "partition/visualize.py", line 75, in <module>
    if (len(pred_red) != len(components)):
TypeError: len() of unsized object

Xiaofei-Kevin-Yang, 2021-05-27:
> > Ok the problem is that you are asking to output tmao room in area1 from the fold 5. Change either the `--res_file` or the `--file_path` so that the target room is indeed in the res_file.
> 
> Hi,I also meet this problem,can you hep me?
> File "partition/visualize.py", line 75, in
> if (len(pred_red) != len(components)):
> TypeError: len() of unsized object

Hi, I also met this issue, did you figure it out?