loicland/superpoint_graph #121: Why train accuracy is smaller than test accuracy ?

hagianga21, 2019-04-30:
Hi Loic,
Why train accuracy is smaller than test accuracy? I am so surprised when I run your validate code as `CUDA_VISIBLE_DEVICES=1 python learning/main.py --dataset sema3d --SEMA3D_PATH 'datasets/semantic3d' --epochs 450 --lr_steps '[350, 400]' --test_nth_epoch 100 \
--model_config 'gru_10,f_8' --ptn_nfeat_stn 11 --nworkers 2 --odir "results/3dseman/standard" --batch_size 1`

loicland, 2019-05-02:
It's not that weird, SPG has very few parameters, and the validation set is small.

You can always try to add more epochs but I an not sure it will increase the performance on the server-side test set.

hagianga21, 2019-05-09:
Well, thank you, it seems magical to me because I 've never seen this before.

closed 2019-05-15 by hagianga21