loicland/superpoint_graph #122: Training Problem - I am using Pytorch 0.3.0.post4. What could I check to solver it ?

hagianga21, 2019-04-30:
Epoch 88/450 (results/bremen/standard):
100%|███████████████████████████████████████████████████████████████| 11/11 [00:02<00:00,  4.86it/s]
-> Train accuracy: 74.09456740442656,   Loss: 0.7768359401009299
Epoch 89/450 (results/bremen/standard):
100%|███████████████████████████████████████████████████████████████| 11/11 [00:02<00:00,  4.90it/s]
100%|█████████████████████████████████████████████████████████████████| 4/4 [00:04<00:00,  1.02s/it]
-> Train accuracy: 73.15096251266465,   Loss: 0.7972265373576771,       Test accuracy: 80.00972289742343,     Test oAcc: 0.861372385881529,    Test avgIoU: 0.5091587632388448
Epoch 90/450 (results/bremen/standard):
  9%|█████▊                                                          | 1/11 [00:00<00:06,  1.52it/s]Traceback (most recent call last):
  File "learning/main.py", line 388, in <module>
    main()
  File "learning/main.py", line 286, in main
    acc, loss, oacc, avg_iou = train()
  File "learning/main.py", line 184, in train
    outputs = model.ecc(embeddings)
  File "/home/gtruong/anaconda3/envs/super2/lib/python3.6/site-packages/torch/nn/modules/module.py", line 325, in __call__
    result = self.forward(*input, **kwargs)
  File "/home/gtruong/project/superpoint_graph/learning/graphnet.py", line 97, in forward
    input = module(input)
  File "/home/gtruong/anaconda3/envs/super2/lib/python3.6/site-packages/torch/nn/modules/module.py", line 325, in __call__
    result = self.forward(*input, **kwargs)
  File "/home/gtruong/project/superpoint_graph/learning/modules.py", line 54, in forward
    input = ecc.GraphConvFunction(nc, nc, idxn, idxe, degs, degs_gpu, self._edge_mem_limit)(hx, weights)
  File "/home/gtruong/project/superpoint_graph/learning/ecc/GraphConvModule.py", line 67, in forward
    cuda_kernels.conv_aggregate_fw(output.narrow(0,startd,numd), products.view(-1,self._out_channels), self._degs_gpu.narrow(0,startd,numd))
  File "/home/gtruong/project/superpoint_graph/learning/ecc/cuda_kernels.py", line 118, in conv_aggregate_fw
    csdegs = torch.cumsum(degs,0)
RuntimeError: scan failed to synchronize: an illegal memory access was encountered

Traceback (most recent call last):
  File "cupy/cuda/driver.pyx", line 192, in cupy.cuda.driver.moduleUnload
  File "cupy/cuda/driver.pyx", line 81, in cupy.cuda.driver.check_status
cupy.cuda.driver.CUDADriverError: CUDA_ERROR_ILLEGAL_ADDRESS: an illegal memory access was encountered
Exception ignored in: 'cupy.cuda.function.Module.__dealloc__'
Traceback (most recent call last):
  File "cupy/cuda/driver.pyx", line 192, in cupy.cuda.driver.moduleUnload
  File "cupy/cuda/driver.pyx", line 81, in cupy.cuda.driver.check_status
cupy.cuda.driver.CUDADriverError: CUDA_ERROR_ILLEGAL_ADDRESS: an illegal memory access was encountered
Traceback (most recent call last):
  File "cupy/cuda/driver.pyx", line 192, in cupy.cuda.driver.moduleUnload
  File "cupy/cuda/driver.pyx", line 81, in cupy.cuda.driver.check_status
cupy.cuda.driver.CUDADriverError: CUDA_ERROR_ILLEGAL_ADDRESS: an illegal memory access was encountered
Exception ignored in: 'cupy.cuda.function.Module.__dealloc__'
Traceback (most recent call last):
  File "cupy/cuda/driver.pyx", line 192, in cupy.cuda.driver.moduleUnload
  File "cupy/cuda/driver.pyx", line 81, in cupy.cuda.driver.check_status
cupy.cuda.driver.CUDADriverError: CUDA_ERROR_ILLEGAL_ADDRESS: an illegal memory access was encountered

I am using Pytorch 0.3.0.post4. What could I check to solver it ?

loicland, 2019-05-02:
Could you check the issues #105 and #81 and see if it helps?

If it is the stochastic error that I think it is, the easiest fix is simply to start the training again with `--resume RESUME` option.

closed 2019-05-09 by hagianga21

loicland, 2020-01-16:
better late than ever, this bug is fixed if you use pygeometric implementation of ecc, which doesn't seem to have this issue.

THHHomas, 2020-02-21:
@loicland Where could I get the pygeometric implementation of ecc? or I need to implement myself?