loicland/superpoint_graph #129: How to control the size of each superpoint?

sunshineatnoon, 2019-06-18:
Hi,

Thanks for open-sourcing this awesome project. In my case, I would like to control the size of each super point so that each one of them doesn't contain too many points. I've tried to decrease reg_strength, which does give me more superpoint, but still, I got some superpoints contain too many points in them. Can you please point me if I can control the size of each superpoint? Thanks!

loicland, 2019-06-26:
Hi, sorry for the delay I was travelling.

I pushed a new version of cut pursuit allowing for controlling the maximal size of a superpoint (it will be required anyway for the new version of SPG upcoming in the next few days). Pull a fresh version and recompile `libcp.so`.

`cut_pursuit` now has a `cutoff` option which lets you chose the maximum size for a superpoint. Each point that is too small is greedily merged with a neighbor. Let me know if you encounter any issues.

sunshineatnoon, 2019-06-30:
Thanks so much!

closed 2019-06-30 by sunshineatnoon