loicland/superpoint_graph #130: rgb array scrambled by libply_c.prune()

bimtauer, 2019-06-26:
Thanks for your great work! 

I discovered an issue when trying to visualize pruned point clouds with their original colors:
The rgb array returned by prune is scrambled as visible in the screenshot.

[image]

I could reproduce this behavior with the following lines:

`xyz = np.random.rand(100, 3) * 1000
rgb = np.random.rand(100, 3)
labels = np.random.rand(100, 1)

xyz_pr, rgb_pr, labels_pr = libply_c.prune(xyz, 0.03, rgb, labels, 1)
`

loicland, 2019-06-26:
Hi,

what do you mean by scramble?

Eitherway, can you try:
```
xyz = np.ascontiguousarray(xyz, dtype='float32')
rgb = np.ascontiguousarray(rgb, dtype='uint8')
```

bimtauer, 2019-06-26:
Hi there, thanks for the quick answer! 

Adding that line to read_las function befor pruning did the job!

closed 2019-06-26 by bimtauer