loicland/superpoint_graph #131: Some problems about compile

Liu-Feng, 2019-06-30:
Hi, thanks for this open this wonderful projects.
 my os is **macOS** 10.14.5 and python **3.6.8**.  
**I replace the libpython3.6..so with libpython3.6.a**
**Question1:**
I countered some obstacles in compiling ply_c
$cmake ...
output:(spg) 192:ply_c liufeng$ cmake . -DPYTHON_LIBRARY=$/Users/liufeng/anaconda3/envs/spg/lib/libpython3.6m.a -DPYTHON_INCLUDE_DIR=$/Users/liufeng/anaconda3/envs/spg/include/python3.6m -DBOOST_INCLUDEDIR=$/Users/liufeng/anaconda3/envs/spg/include -DEIGEN3_INCLUDE_DIR=$/Users/liufeng/anaconda3/envs/spg/include/eigen3
-- The C compiler identification is AppleClang 10.0.1.10010046
-- The CXX compiler identification is AppleClang 10.0.1.10010046
-- Check for working C compiler: /Library/Developer/CommandLineTools/usr/bin/cc
-- Check for working C compiler: /Library/Developer/CommandLineTools/usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /Library/Developer/CommandLineTools/usr/bin/c++
-- Check for working CXX compiler: /Library/Developer/CommandLineTools/usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Boost version: 1.67.0
-- Found the following Boost libraries:
--   python
--   numpy
Boost includes ARE /Users/liufeng/anaconda3/envs/spg/include
Boost LIBRARIES ARE /Users/liufeng/anaconda3/envs/spg/lib
PYTHON LIBRARIES ARE /Users/liufeng/pc_Python/superpoint_graph/partition/ply_c/$/Users/liufeng/anaconda3/envs/spg/lib/libpython3.6m.a
-- Configuring done
-- Generating done
-- Build files have been written to: /Users/liufeng/pc_Python/superpoint_graph/partition/ply_c

****$make
output:
[ 50%] Building CXX object CMakeFiles/ply_c.dir/ply_c.cpp.o
clang: error: unsupported option '-fopenmp'
make[2]: *** [CMakeFiles/ply_c.dir/ply_c.cpp.o] Error 1
make[1]: *** [CMakeFiles/ply_c.dir/all] Error 2
make: *** [all] Error 2****
I couldn't make sense this error. Can you give me a hand in your convenient. Did anyone else met this error. Thanks.

**Question2:**
The Cmake version is  3.14.0, you have said that the Cmake version is at least 3.5 in issue 13 but the latest version is 3.15? Did I go to wrong address.

loicland, 2019-07-01:
Hi,

I don't have access to a mac to test compatibility, but it seems that the default compiler of mac doesn't support openMP.

This [issue](https://github.com/alexdobin/STAR/issues/375) seems to adress it.

Let me know if you solved it, and please post what ended up working for you.

Thanks.

Liu-Feng, 2019-07-03:
> Hi,
> 
> I don't have access to a mac to test compatibility, but it seems that the default compiler of mac doesn't support openMP.
> 
> This [issue](https://github.com/alexdobin/STAR/issues/375) seems to adress it.
> 
> Let me know if you solved it, and please post what ended up working for you.
> 
> Thanks.

closed 2019-07-03 by Liu-Feng

mpelchat04, 2020-02-25:
@Liu-Feng Have you found a workaround for the `make` error? I get similar error...
```
make
make[2]: *** No rule to make target /lib/libpython3.6m.so', needed by libply_c.so'. Stop.
make[1]: *** [CMakeFiles/ply_c.dir/all] Error 2
make: *** [all] Error 2
```  
cmake ran well, without warnings nor errors.

loicland, 2020-02-26:
Hi,

are you sure you are using pyhton3.6m and not another version? If so you need to adapt it in the comand line.