loicland/superpoint_graph #132: problems when compiling libcp

colordown, 2019-07-01:
Dear Ioicland, I come come across some problems when I compile the libcp library(ps: I have successfully compiled libply_c).

/home/ljw/superpoint_graph/partition/cut-pursuit/src/cutpursuit.cpp:90:80: error: no matching function for call to ‘cut_pursuit(const uint32_t&, const uint32_t&, const uint32_t&, const float*&, const uint32_t*&, const uint32_t*&, const float*&, __gnu_cxx::__alloc_traits<std::allocator<float> >::value_type*, float*, std::vector<unsigned int>&, std::vector<std::vector<unsigned int> >&, float&, float, float, float)’
              , solution.data(), in_component, components, lambda, 1.f, 2.f, 2.f);
In file included from /home/ljw/superpoint_graph/partition/cut-pursuit/src/cutpursuit.cpp:9:0:
/home/ljw/superpoint_graph/partition/cut-pursuit/src/../include/../include/API.h:183:6: note: candidate: template<class T> void CP::cut_pursuit(uint32_t, uint32_t, uint32_t, const T*, const uint32_t*, const uint32_t*, const T*, const T*, T*, T, uint32_t, T, T, float)
 void cut_pursuit(const uint32_t n_nodes, const uint32_t n_edges, const uint32_t nObs
      ^
/home/ljw/superpoint_graph/partition/cut-pursuit/src/../include/../include/API.h:183:6: note:   template argument deduction/substitution failed:
/home/ljw/superpoint_graph/partition/cut-pursuit/src/cutpursuit.cpp:90:80: note:   cannot convert ‘in_component’ (type ‘std::vector<unsigned int>’) to type ‘float’
              , solution.data(), in_component, components, lambda, 1.f, 2.f, 2.f);
...

CMakeFiles/cp.dir/build.make:62: recipe for target 'CMakeFiles/cp.dir/cutpursuit.cpp.o' failed
make[2]: *** [CMakeFiles/cp.dir/cutpursuit.cpp.o] Error 1
CMakeFiles/Makefile2:67: recipe for target 'CMakeFiles/cp.dir/all' failed
make[1]: *** [CMakeFiles/cp.dir/all] Error 2
Makefile:83: recipe for target 'all' failed
make: *** [all] Error 2

loicland, 2019-07-01:
Hi,

can you tell me the commit version of your cut-pursuit repository please?

colordown, 2019-07-01:
Thanks for your responding,
I git clone it from https://github.com/loicland/cut-pursuit before your commits on Jun 28, 2019. Is the problem related to your recent change?

loicland, 2019-07-01:
Yes, can you clone a fresh repository of cut pursuit? I updated the compilation command as well.

colordown, 2019-07-01:
I tried it, it seems that your new command has some problem.
why do you need to make a new dir called build?
CMake Error: The source directory "/home/ljw/superpoint_graph/partition/cut-pursuit/build" does not appear to contain CMakeLists.txt.

loicland, 2019-07-01:
My bad, try it now with the updated command line (`..` instead of `.`).

colordown, 2019-07-01:
(o゜▽゜)o☆[BINGO!] It works now! Thanks again for your reply. I am now go on running your code!

closed 2019-07-03 by colordown

atineoSE, 2019-07-17:
@loicland I have the same problem reported by @colordown but I can't seem to fix it with the steps above.

If I build as per the updated README instructions over the commit referred to by this repo (6e6ee14) I get this error (same as @colordown initially):
```
my_path/superpoint_graph/partition/cut-pursuit/src/cutpursuit.cpp: In function ‘PyObject* cutpursuit_seg(const boost::python::numpy::ndarray&, const boost::python::numpy::ndarray&, const boost::python::numpy::ndarray&, const boost::python::numpy::ndarray&, float)’:
my_path/superpoint_graph/partition/cut-pursuit/src/cutpursuit.cpp:90:80: error: no matching function for call to ‘cut_pursuit(const uint32_t&, const uint32_t&, const uint32_t&, const float*&, const uint32_t*&, const uint32_t*&, const float*&, __gnu_cxx::__alloc_traits<std::allocator<float> >::value_type*, float*, std::vector<unsigned int>&, std::vector<std::vector<unsigned int> >&, float&, float, float, float)’
              , solution.data(), in_component, components, lambda, 1.f, 2.f, 2.f);
                                                                                ^
my_path/superpoint_graph/partition/cut-pursuit/src/cutpursuit.cpp:90:80: note: candidates are:
In file included from my_path/superpoint_graph/partition/cut-pursuit/src/cutpursuit.cpp:9:0:
my_path/superpoint_graph/partition/cut-pursuit/src/../include/../include/API.h:183:6: note: template<class T> void CP::cut_pursuit(uint32_t, uint32_t, uint32_t, const T*, const uint32_t*, const uint32_t*, const T*, const T*, T*, T, uint32_t, T, T, float)
 void cut_pursuit(const uint32_t n_nodes, const uint32_t n_edges, const uint32_t nObs
      ^
my_path/superpoint_graph/partition/cut-pursuit/src/../include/../include/API.h:183:6: note:   template argument deduction/substitution failed:
my_path/superpoint_graph/partition/cut-pursuit/src/cutpursuit.cpp:90:80: note:   cannot convert ‘in_component’ (type ‘std::vector<unsigned int>’) to type ‘float’
              , solution.data(), in_component, components, lambda, 1.f, 2.f, 2.f);

[...]

make[2]: *** [CMakeFiles/cp.dir/cutpursuit.cpp.o] Error 1
make[1]: *** [CMakeFiles/cp.dir/all] Error 2
make: *** [all] Error 2

```

(Minor note,  there is a missing`/src` in the build command path, it should read as follows: `cmake ../src -DPYTHON_LIBRARY=...`)

If I update the cut-pursuit submodule to master, then I get an error in the configuration phase with cmake as follows:

```
$ cd ..
$ rm -rf build
$ git status
HEAD detached at 6e6ee14
nothing to commit, working directory clean

$ git checkout master

$ mkdir build
$ cd build
$ cmake ../src -DPYTHON_LIBRARY=$CONDAENV/lib/libpython3.7m.a -DPYTHON_INCLUDE_DIR=$CONDAENV/include/python3.7m -DBOOST_INCLUDEDIR=$CONDAENV/include
-- The C compiler identification is GNU 4.8.5
-- The CXX compiler identification is GNU 4.8.5
-- Checking whether C compiler has -isysroot
-- Checking whether C compiler has -isysroot - yes
-- Checking whether C compiler supports OSX deployment target flag
-- Checking whether C compiler supports OSX deployment target flag - yes
-- Check for working C compiler: /Users/adriantineo/anaconda3/envs/pytorchNoCUDA/bin/cc
-- Check for working C compiler: /Users/adriantineo/anaconda3/envs/pytorchNoCUDA/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Checking whether CXX compiler has -isysroot
-- Checking whether CXX compiler has -isysroot - yes
-- Checking whether CXX compiler supports OSX deployment target flag
-- Checking whether CXX compiler supports OSX deployment target flag - yes
-- Check for working CXX compiler: /Users/adriantineo/anaconda3/envs/pytorchNoCUDA/bin/c++
-- Check for working CXX compiler: /Users/adriantineo/anaconda3/envs/pytorchNoCUDA/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Found PythonLibs: /Users/adriantineo/anaconda3/envs/pytorchNoCUDA/lib/libpython3.7m.a (found version "3.7.3") 
-- Found PythonInterp: /Users/adriantineo/anaconda3/envs/pytorchNoCUDA/bin/python3.7 (found version "3.7.3") 
CMake Error at CMakeLists.txt:7 (include):
  include could not find load file:

    FindNumPy

-- Boost version: 1.67.0
-- Found the following Boost libraries:
--   graph
-- Boost version: 1.67.0
-- Found the following Boost libraries:
--   numpy37
--   python37
PYTHON LIBRARIES ARE /Users/adriantineo/anaconda3/envs/pytorchNoCUDA/lib/libpython3.7m.a
/Users/adriantineo/anaconda3/envs/pytorchNoCUDA/lib/libboost_numpy37.dylib/Users/adriantineo/anaconda3/envs/pytorchNoCUDA/lib/libboost_python37.dylib
CMake Warning (dev) in CMakeLists.txt:
  No cmake_minimum_required command is present.  A line of code such as

    cmake_minimum_required(VERSION 3.14)

  should be added at the top of the file.  The version specified may be lower
  if you wish to support older CMake versions for this project.  For more
  information run "cmake --help-policy CMP0000".
This warning is for project developers.  Use -Wno-dev to suppress it.

-- Configuring incomplete, errors occurred!
```

atineoSE, 2019-07-19:
Following up on my previous message @loicland, I have tried several commits from the cut-pursuit submodule but I can't get it to compile successfully. 

I was initially using python 3.7, boost 1.67 and cmake 3.14. For the sake of keeping things simple, I have changed to the suggested python 3.6, boost 1.65.1 and cmake 3.5.

For completeness, here is an account of all problems. 

Commit 6e6ee14, which is the one linked from master in this repo -> "no matching function" error:
```
(supergraphNoCUDA) imacAdrian:build adriantineo$ cmake ../src -DPYTHON_LIBRARY=$CONDAENV/lib/libpython3.6m.a -DPYTHON_INCLUDE_DIR=$CONDAENV/include/python3.6m -DBOOST_INCLUDEDIR=$CONDAENV/include
-- The C compiler identification is GNU 4.8.5
-- The CXX compiler identification is GNU 4.8.5
-- Checking whether C compiler has -isysroot
-- Checking whether C compiler has -isysroot - yes
-- Checking whether C compiler supports OSX deployment target flag
-- Checking whether C compiler supports OSX deployment target flag - yes
-- Check for working C compiler: /Users/adriantineo/anaconda3/envs/supergraphNoCUDA/bin/cc
-- Check for working C compiler: /Users/adriantineo/anaconda3/envs/supergraphNoCUDA/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Checking whether CXX compiler has -isysroot
-- Checking whether CXX compiler has -isysroot - yes
-- Checking whether CXX compiler supports OSX deployment target flag
-- Checking whether CXX compiler supports OSX deployment target flag - yes
-- Check for working CXX compiler: /Users/adriantineo/anaconda3/envs/supergraphNoCUDA/bin/c++
-- Check for working CXX compiler: /Users/adriantineo/anaconda3/envs/supergraphNoCUDA/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
CMake Warning at /Users/adriantineo/anaconda3/envs/supergraphNoCuda/share/cmake-3.5/Modules/FindBoost.cmake:725 (message):
  Imported targets not available for Boost version 106501
Call Stack (most recent call first):
  /Users/adriantineo/anaconda3/envs/supergraphNoCuda/share/cmake-3.5/Modules/FindBoost.cmake:762 (_Boost_COMPONENT_DEPENDENCIES)
  /Users/adriantineo/anaconda3/envs/supergraphNoCuda/share/cmake-3.5/Modules/FindBoost.cmake:1309 (_Boost_MISSING_DEPENDENCIES)
  CMakeLists.txt:17 (find_package)

CMake Warning at /Users/adriantineo/anaconda3/envs/supergraphNoCuda/share/cmake-3.5/Modules/FindBoost.cmake:725 (message):
  Imported targets not available for Boost version 106501
Call Stack (most recent call first):
  /Users/adriantineo/anaconda3/envs/supergraphNoCuda/share/cmake-3.5/Modules/FindBoost.cmake:762 (_Boost_COMPONENT_DEPENDENCIES)
  /Users/adriantineo/anaconda3/envs/supergraphNoCuda/share/cmake-3.5/Modules/FindBoost.cmake:1309 (_Boost_MISSING_DEPENDENCIES)
  CMakeLists.txt:17 (find_package)

-- Boost version: 1.65.1
-- Found the following Boost libraries:
--   graph
--   numpy
-- Found PythonLibs: /Users/adriantineo/anaconda3/envs/supergraphNoCUDA/lib/libpython3.6m.a (found version "3.6.8") 
PYTHON LIBRARIES ARE /Users/adriantineo/anaconda3/envs/supergraphNoCUDA/lib/libpython3.6m.a
-- Configuring done
-- Generating done
-- Build files have been written to: myPath/superpoint_graph/partition/cut-pursuit/build
(supergraphNoCUDA) imacAdrian:build adriantineo$ make
Scanning dependencies of target cp
[ 50%] Building CXX object CMakeFiles/cp.dir/cutpursuit.cpp.o
myPath/superpoint_graph/partition/cut-pursuit/src/cutpursuit.cpp: In function ‘PyObject* cutpursuit_seg(const boost::python::numpy::ndarray&, const boost::python::numpy::ndarray&, const boost::python::numpy::ndarray&, const boost::python::numpy::ndarray&, float)’:
myPath/superpoint_graph/partition/cut-pursuit/src/cutpursuit.cpp:90:80: error: no matching function for call to ‘cut_pursuit(const uint32_t&, const uint32_t&, const uint32_t&, const float*&, const uint32_t*&, const uint32_t*&, const float*&, __gnu_cxx::__alloc_traits<std::allocator<float> >::value_type*, float*, std::vector<unsigned int>&, std::vector<std::vector<unsigned int> >&, float&, float, float, float)’
              , solution.data(), in_component, components, lambda, 1.f, 2.f, 2.f);

[...]

              make[2]: *** [CMakeFiles/cp.dir/cutpursuit.cpp.o] Error 1
make[1]: *** [CMakeFiles/cp.dir/all] Error 2
make: *** [all] Error 2
```

Commit e9501d1, which is the current tip of the master branch -> "could not find FindNumPy" error:
```
[... 115 lines cut ...]
Scanning dependencies of target cp
[ 50%] Building CXX object CMakeFiles/cp.dir/cutpursuit.cpp.o

[...]

[100%] Linking CXX shared library libcp.dylib
undef: boost::python::objects::py_function_impl_base::max_arity() const
undef: typeinfo for boost::python::objects::py_function_impl_base
undef: boost::python::converter::registry::lookup(boost::python::type_info)

[...]

  "boost::python::converter::registry::insert(_object* (*)(void const*), boost::python::type_info, _typeobject const* (*)())", referenced from:
      init_module_libcp()     in cutpursuit.cpp.o
ld: symbol(s) not found for architecture x86_64
collect2: error: ld returned 1 exit status
make[2]: *** [libcp.dylib] Error 1
make[1]: *** [CMakeFiles/cp.dir/all] Error 2
make: *** [all] Error 2

```

Since I changed from cmake 3.14 to cmake 3.5, new warnings appear related to the boost library. I have tried some combinations of cmake and boost, as suggested [here](https://stackoverflow.com/a/42124857/1811775), but I always get the same warnings with cmake, then the linking error.  

To sum up, I can't find any combination of commit and dependencies that will compile and link successfully. Any help much appreciated, thanks.

loicland, 2019-07-19:
Hi,

make sure that you have the latest commit e9501d of cut-pursuit instead of 6e6ee14.

I also just pushed a new update on the new default branch ssp+spg. Let me know if you still have the same problem.

atineoSE, 2019-07-19:
Yes @loicland, I have tried with the latest commit in cut-pursuit (e9501d), I get a `include could not find load file: FindNumPy` error, as detailed above. As I understand, compiling this submodule is independent of the new content in the ssp+spg branch, correct?

loicland, 2019-07-19:
Hi,

I don't have access to my work station right now, but could you try to comment line 7 of cut-pursuit/src/CMakeList.txt?

Some modification made to the cut pursuit repo are necessary for supervised_partition to work.

atineoSE, 2019-07-20:
Thanks for the fast response @loicland. I still cannot build, unfortunately. Please, see below for details. 

It seems like the boost libraries can't be found properly (warning: `Imported targets not available for Boost version 106700` in cmake, then error when building). I have tried to look for help on this, but could not solve it.

Summary:
- I am now trying with commit e9501d1 in the cut-pursuit submodule (current tip of the master branch)
- I have commented out line 7 as suggested (`include(FindNumPy)`) -> this makes the cmake error go away
- I updated from boost 1.65 to 1.67, since this seems to be supported, as per the `CMakeLists.txt` file
- I am using python 3.6 and cmake 3.5
- I am using GCC from the conda distribution as compiler (GNU 4.8.5)

```
(supergraphNoCUDA) imacAdrian:build adriantineo$ export CONDAENV=/Users/adriantineo/anaconda3/envs/supergraphNoCUDA
(supergraphNoCUDA) imacAdrian:build adriantineo$ cmake ../src -DPYTHON_LIBRARY=$CONDAENV/lib/libpython3.6m.a -DPYTHON_INCLUDE_DIR=$CONDAENV/include/python3.6m -DBOOST_INCLUDEDIR=$CONDAENV/include
-- The C compiler identification is GNU 4.8.5
-- The CXX compiler identification is GNU 4.8.5
-- Checking whether C compiler has -isysroot
-- Checking whether C compiler has -isysroot - yes
-- Checking whether C compiler supports OSX deployment target flag
-- Checking whether C compiler supports OSX deployment target flag - yes
-- Check for working C compiler: /Users/adriantineo/anaconda3/envs/supergraphNoCUDA/bin/cc
-- Check for working C compiler: /Users/adriantineo/anaconda3/envs/supergraphNoCUDA/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Checking whether CXX compiler has -isysroot
-- Checking whether CXX compiler has -isysroot - yes
-- Checking whether CXX compiler supports OSX deployment target flag
-- Checking whether CXX compiler supports OSX deployment target flag - yes
-- Check for working CXX compiler: /Users/adriantineo/anaconda3/envs/supergraphNoCUDA/bin/c++
-- Check for working CXX compiler: /Users/adriantineo/anaconda3/envs/supergraphNoCUDA/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Found PythonLibs: /Users/adriantineo/anaconda3/envs/supergraphNoCUDA/lib/libpython3.6m.a (found version "3.6.8") 
-- Found PythonInterp: /Users/adriantineo/anaconda3/envs/supergraphNoCUDA/bin/python3.6 (found version "3.6.8") 
CMake Warning at /Users/adriantineo/anaconda3/envs/supergraphNoCuda/share/cmake-3.5/Modules/FindBoost.cmake:725 (message):
  Imported targets not available for Boost version 106700
Call Stack (most recent call first):
  /Users/adriantineo/anaconda3/envs/supergraphNoCuda/share/cmake-3.5/Modules/FindBoost.cmake:762 (_Boost_COMPONENT_DEPENDENCIES)
  /Users/adriantineo/anaconda3/envs/supergraphNoCuda/share/cmake-3.5/Modules/FindBoost.cmake:1309 (_Boost_MISSING_DEPENDENCIES)
  CMakeLists.txt:14 (find_package)

-- Boost version: 1.67.0
-- Found the following Boost libraries:
--   graph
CMake Warning at /Users/adriantineo/anaconda3/envs/supergraphNoCuda/share/cmake-3.5/Modules/FindBoost.cmake:725 (message):
  Imported targets not available for Boost version 106700
Call Stack (most recent call first):
  /Users/adriantineo/anaconda3/envs/supergraphNoCuda/share/cmake-3.5/Modules/FindBoost.cmake:762 (_Boost_COMPONENT_DEPENDENCIES)
  /Users/adriantineo/anaconda3/envs/supergraphNoCuda/share/cmake-3.5/Modules/FindBoost.cmake:1309 (_Boost_MISSING_DEPENDENCIES)
  CMakeLists.txt:19 (find_package)

-- Boost version: 1.67.0
-- Found the following Boost libraries:
--   numpy36
PYTHON LIBRARIES ARE /Users/adriantineo/anaconda3/envs/supergraphNoCUDA/lib/libpython3.6m.a
/Users/adriantineo/anaconda3/envs/supergraphNoCUDA/lib/libboost_numpy36.dylib
CMake Warning (dev) in CMakeLists.txt:
  No cmake_minimum_required command is present.  A line of code such as

    cmake_minimum_required(VERSION 3.5)

  should be added at the top of the file.  The version specified may be lower
  if you wish to support older CMake versions for this project.  For more
  information run "cmake --help-policy CMP0000".
This warning is for project developers.  Use -Wno-dev to suppress it.

-- Configuring done
-- Generating done
-- Build files have been written to: myPath/superpoint_graph/partition/cut-pursuit/build
(supergraphNoCUDA) imacAdrian:build adriantineo$ make
Scanning dependencies of target cp
[ 50%] Building CXX object CMakeFiles/cp.dir/cutpursuit.o
In file included from myPath/superpoint_graph/partition/cut-pursuit/src/../include/../include/CutPursuit.h:2:0,
                 from myPath/superpoint_graph/partition/cut-pursuit/src/../include/../include/CutPursuit_L2.h:2,
                 from myPath/superpoint_graph/partition/cut-pursuit/src/../include/../include/API.h:3,
                 from myPath/superpoint_graph/partition/cut-pursuit/src/cutpursuit.cpp:9:
myPath/superpoint_graph/partition/cut-pursuit/src/../include/../include/Graph.h:48:1: error: expected unqualified-id before ‘using’
 using Graph = typename boost::adjacency_list<boost::vecS, boost::vecS, boost::directedS, VertexAttribute<T>, EdgeAttribute<T> >;
 ^
 [...]

 myPath/superpoint_graph/partition/cut-pursuit/src/../include/../include/CutPursuit.h:400:5: error: expected ‘}’ at end of input
make[2]: *** [CMakeFiles/cp.dir/cutpursuit.o] Error 1
make[1]: *** [CMakeFiles/cp.dir/all] Error 2
make: *** [all] Error 2
```

loicland, 2019-07-22:
Hi,

are you able to compile ply_c.o correctly?

The warning: "Imported targets not available for Boost version 106700" seems to imply that you would require more recent version of Cmake (see [this thread](https://stackoverflow.com/questions/42123509/cmake-finds-boost-but-the-imported-targets-not-available-for-boost-version) ), but I compiled it on all three of my stations with 3.5.1. So I doubt that's the problem.

The compilation error implies on the other hand that your GCC is too old, as the `using` causing the error is >c++11 only. I have tested the code with GCC6.5.0, and 5.4.0. Maybe you could upgrade it and report here the results?

atineoSE, 2019-07-22:
Thanks for the follow-up, @loicland. However, I still cannot build the cutpoint library successfully.

Yes, I have built ply_c successfully already. 

I have updated to cmake 3.12, which is the version matching boost 1.67, as per the thread you linked to. That makes the `Imported targets not available` warning go away.

I could [update to gcc 5.2](https://anaconda.org/psi4/gcc-5), which is the highest gcc version precompiled for macOS/OSX in conda, as seen [here](https://anaconda.org/search?q=gcc) at the time of this writing.

Good tip on the C++11 standard. I could get rid of the "expected unqualified-id before ‘using’" error by adding the following line to the file `CMakeLists.txt`:
```
SET(CMAKE_CXX_FLAGS "${CMAKE_CXX_FLAGS} -std=c++11")
```

Still, the library won't link:
```
[100%] Linking CXX shared library libcp.dylib
undef: boost::python::objects::function_object(boost::python::objects::py_function const&, std::pair<boost::python::detail::keyword const*, boost::python::detail::keyword const*> const&)
Undefined symbols for architecture x86_64:
  "boost::python::objects::function_object(boost::python::objects::py_function const&, std::pair<boost::python::detail::keyword const*, boost::python::detail::keyword const*> const&)", referenced from:
      boost::python::api::object boost::python::detail::make_function_aux<_object* (*)(boost::python::numpy::ndarray const&, boost::python::numpy::ndarray const&, boost::python::numpy::ndarray const&, boost::python::numpy::ndarray const&, float, int, int, float), boost::python::default_call_policies, boost::mpl::vector9<_object*, boost::python::numpy::ndarray const&, boost::python::numpy::ndarray const&, boost::python::numpy::ndarray const&, boost::python::numpy::ndarray const&, float, int, int, float>, mpl_::int_<3> >(_object* (*)(boost::python::numpy::ndarray const&, boost::python::numpy::ndarray const&, boost::python::numpy::ndarray const&, boost::python::numpy::ndarray const&, float, int, int, float), boost::python::default_call_policies const&, boost::mpl::vector9<_object*, boost::python::numpy::ndarray const&, boost::python::numpy::ndarray const&, boost::python::numpy::ndarray const&, boost::python::numpy::ndarray const&, float, int, int, float> const&, std::pair<boost::python::detail::keyword const*, boost::python::detail::keyword const*> const&, mpl_::int_<3>) in cutpursuit.o
ld: symbol(s) not found for architecture x86_64
collect2: error: ld returned 1 exit status
make[2]: *** [libcp.dylib] Error 1
make[1]: *** [CMakeFiles/cp.dir/all] Error 2
make: *** [all] Error 2
```

loicland, 2019-07-22:
What version of XCode are you using?

atineoSE, 2019-07-22:
For reference, I am using Xcode 10.2. However, I believe by using the conda environment and the conda installed gcc, I am effectively bypassing Xcode tools. In fact, Xcode's gcc, which is based in clang, won't work because it doesn't support OpenMP. The brew installed gcc won't work either, because latest builds don't support OpenMP either, and brew does no longer accept the --without-multilib option, which enabled OpenMP. It seems like gcc is my best option so far in macOS/OSX.

loicland, 2019-07-22:
I am not well-versed at all in macOS, and don't have access to one such device to test things out myself sadly.

These two threads ([1](https://forum.qt.io/topic/20654/solved-undefined-symbols-for-architecture-x86_64-when-building-hidapi/12) [2](https://github.com/kmagiera/react-native-gesture-handler/issues/387)) address a similar problem. Seems like an I/O Kit linkage issue.

Do let me know if you manage to solve this issue, or I'll try to reach out to macOS users for help.

atineoSE, 2019-07-22:
Thanks for the tip @loicland. However, I don't think those links work, the first reports about a different library, and the second is for a project built within Xcode. I am in a conda environment using a conda-installed compiler and libraries, so not using Xcode Developer tools, to the best of my knowledge. After days invested in building this library, I am running out of ideas to try.

loicland, 2019-07-22:
How about rebuilding boost with c++11 support [like suggested here](https://stackoverflow.com/questions/18389799/boost-python-undefined-symbols-error-even-though-the-lib-files-are-linked)?

Sadly I can't help you more as I don't have a mac station. Will ask around.

atineoSE, 2019-07-22:
That's a good tip. Unfortunately, brew [no longer accepts formulas](https://github.com/Homebrew/homebrew-core/issues/31510), which leaves [a lot of users](https://discourse.brew.sh/t/maintaining-a-tap-to-provide-options-for-a-formula/3871) with no good options but to fetch sources and build locally for each dependency not using defaults.

atineoSE, 2019-07-22:
I know the libraries are there, installed by conda, I just can't get cmake link to them. It looks like [this solution](https://stackoverflow.com/a/42276376/1811775) should work, but it doesn't, same linking error.

```
undef: boost::python::objects::function_object(boost::python::objects::py_function const&, std::pair<boost::python::detail::keyword const*, boost::python::detail::keyword const*> const&)
Undefined symbols for architecture x86_64:
```

nicolas-chaulet, 2019-07-22:
I am having a similar issue and I think that the problem comes from the optional parameters in the python bindings. I could not figure out why nor how to fix it though...

atineoSE, 2019-07-22:
What errors are you seeing @nicolas-chaulet? Beyond the problems reported here, I've had to sift through dozens of additional problems. Maybe I can help you. Are you building in macOS as well?

nicolas-chaulet, 2019-07-22:
Ths same as the last one above, it fails to link because it does not find some symbols in the boost library.

loicland, 2019-07-24:
Sadly I don't see any easy way to solve this for the time being.

We are planning to interface [this new implementation](https://github.com/1a7r0ch3/parallel-cut-pursuit) of cut pursuit with SPG in the next month or so. since it doesn't rely on boost it could solve your issue.

I will keep you posted.

atineoSE, 2019-07-24:
Thanks a lot @loicland, I'd be interested to try it when it's available. At the moment, I have given up on trying to build on macOS.

loicland, 2019-07-25:
@atineoSE  could you confirm that @nicolas-chaulet 's fix allows you to compile cut pursuit?

atineoSE, 2019-07-26:
@loicland thanks for the heads-up and @nicolas-chaulet for his changes. I could finally compile and link! 🥳

However, I had to change things around a bit. Let me break down the process. This is all over @nicolas-chaulet's PR.

For this successful build I installed the following through a conda environment over python 3.7:
- boost 1.67
- cmake 3.12

I am using Apple's Clang from Xcode 10.2 (Apple LLVM version 10.0.1).

These are the errors/warning I got and how I got rid of them:
- cmake error `include could not find load file: FindNumPy`- > Comment out line `include(FindNumPy)` in `src/CMakeLists.txt`
- cmake warning `CMake Warning (dev) in CMakeLists.txt:  No cmake_minimum_required command is present.` -> Add line `cmake_minimum_required(VERSION 3.12)` in `src/CMakeLists.txt`
- cmake warning `MACOSX_RPATH is not specified for the following targets: cp` -> set environment variable as `export MACOSX_RPATH=$CONDAENV/lib` and add line `set(CMAKE_MACOSX_RPATH 0)` to `src/CMakeLists.txt`
- make error `fatal error: 'numpy/ndarrayobject.h' file not found`-> set symbolic link to python site-package with `sudo ln -s $CONDAENV/lib/python3.7/site-packages/numpy/core/include/numpy $CONDAENV/include/numpy`
- make error `error: a space is required between consecutive right angle brackets` -> specific support for C++11 is needed, thus adding the line `SET(CMAKE_CXX_FLAGS "${CMAKE_CXX_FLAGS} -std=c++11")` to `src/CMakeLists.txt`

I haven't run the code yet though, as I still have problems with pytorch, but that is another story...

longmalongma, 2019-10-12:
> the updated command line (`..` instead of `.`).

hi,what meaning of the updated command line (.. instead of .)?

longmalongma, 2019-10-12:
> My bad, try it now with the updated command line (`..` instead of `.`).

hi,what meaning of the updated command line (.. instead of .)?