loicland/superpoint_graph #136: Problem about compile

Liu-Feng, 2019-07-03:
Thanks for opening

Did anyone meet this problems:
My os is Ubuntu14.04 runing in Vmware, and python version is 3.6.8.

$ cmake . **-DPYTHON_LIBRARY=$/home/liufeng/anaconda3/lib/libpython3.6m.so** -DPYTHON_INCLUDE_DIR=$/home/liufeng/anaconda3/include/python3.6m -DBOOST_INCLUDEDIR=$/home/liufeng/anaconda3/include -DEIGEN3_INCLUDE_DIR=$/home/liufeng/anaconda3/include/eigen3

-- Boost version: 1.67.0
-- Found the following Boost libraries:
--   python
--   numpy
Boost includes ARE /home/liufeng/anaconda3/include
Boost LIBRARIES ARE /home/liufeng/anaconda3/lib
-- Found PythonLibs: /home/liufeng/Desktop/superpoint_graph/partition/ply_c/$/home/liufeng/anaconda3/lib/libpython3.6m.so  
**PYTHON LIBRARIES ARE /home/liufeng/Desktop/superpoint_graph/partition/ply_c/$/home/liufeng/anaconda3/lib/libpython3.6m.so**
-- Configuring done
-- Generating done
-- Build files have been written to: /home/liufeng/Desktop/superpoint_graph/partition/ply_c

$make
**make[2]: *** No rule to make target `home/liufeng/anaconda3/lib/libpython3.6m.so', needed by `libply_c.so'.  Stop.**
make[1]: *** [CMakeFiles/ply_c.dir/all] Error 2
make: *** [all] Error 2

And I searched libpython3.6m.so in my computer:
$locate anaconda3/lib/libpython

/home/liufeng/anaconda3/lib/libpython3.6m.a
**/home/liufeng/anaconda3/lib/libpython3.6m.so**
/home/liufeng/anaconda3/lib/libpython3.6m.so.1
/home/liufeng/anaconda3/lib/libpython3.6m.so.1.0

**What confuses me is that I do have libpython3.6m.so in '/home/liufeng/anaconda3/lib', why there are no rule to make target  `home/liufeng/anaconda3/lib/libpython3.6m.so'** 
**And in compiling libcp I met the same question.**
**Thanks for anyone who give me some suggestions.**

loicland, 2019-07-03:
I think the problem might come from the $. Try:
```
CONDAENV=/home/liufeng/anaconda3
cd partition/ply_c
cmake . -DPYTHON_LIBRARY=$CONDAENV/lib/libpython3.6m.so -DPYTHON_INCLUDE_DIR=$CONDAENV/include/python3.6m -DBOOST_INCLUDEDIR=$CONDAENV/include -DEIGEN3_INCLUDE_DIR=$CONDAENV/include/eigen3
make
cd ..
cd cut-pursuit
mkdir build
cd build
cmake .. -DPYTHON_LIBRARY=$CONDAENV/lib/libpython3.6m.so -DPYTHON_INCLUDE_DIR=$CONDAENV/include/python3.6m -DBOOST_INCLUDEDIR=$CONDAENV/include -DEIGEN3_INCLUDE_DIR=$CONDAENV/include/eigen3
make
```

Liu-Feng, 2019-07-04:
Thanks for your help! It works well now. Thank you very much!!!

closed 2019-07-04 by loicland