loicland/superpoint_graph #138: cmake error in ply_c

shobhitjdpsr, 2019-07-24:
[image]
[image]
[image]
[image]

While running cmake . -DPYTHON_LIBRARY=$CONDAENV/lib/libpython3.6m.so -DPYTHON_INCLUDE_DIR=$CONDAENV/include/python3.6m -DBOOST_INCLUDEDIR=$CONDAENV/include -DEIGEN3_INCLUDE_DIR=$CONDAENV/include/eigen3

I am getting an error that boost is too old, despite of installing boost 1.66.

What all can I do to make this work?

shobhitjdpsr, 2019-07-25:
@loicland

loicland, 2019-07-25:
Hi,

what did your set CONDAENV to?

shobhitjdpsr, 2019-07-25:
[image]

[image]
 

@loicland   Please have a llook

shobhitjdpsr, 2019-07-25:
@Pandinosaurus  @loicland

loicland, 2019-07-25:
CONDAENV should be more like something like
/home/your_session/anaconda3

shobhitjdpsr, 2019-07-25:
@loicland 

Hi Thanks for input! 

After doing it I am getting further error:-
[image]

shobhitjdpsr, 2019-07-25:
@loicland 

[image]

shobhitjdpsr, 2019-07-25:
@loicland 
[image]

shobhitjdpsr, 2019-07-25:
@loicland 
Never mind I got it working., by removing cmakecache.txt .

But next error is in make
[image]

shobhitjdpsr, 2019-07-25:
In ply_c while doing MAKE, I am getting following error:

[ 50%] Building CXX object CMakeFiles/ply_c.dir/ply_c.cpp.o
/home/spartan/Desktop/superpoint_graph/partition/ply_c/ply_c.cpp:4:23: fatal error: Eigen/Dense: No such file or directory
 #include <Eigen/Dense>
                       ^
compilation terminated.
CMakeFiles/ply_c.dir/build.make:62: recipe for target 'CMakeFiles/ply_c.dir/ply_c.cpp.o' failed
make[2]: *** [CMakeFiles/ply_c.dir/ply_c.cpp.o] Error 1
CMakeFiles/Makefile2:67: recipe for target 'CMakeFiles/ply_c.dir/all' failed
make[1]: *** [CMakeFiles/ply_c.dir/all] Error 2
Makefile:83: recipe for target 'all' failed
make: *** [all] Error 2
(my_env) spartan@spartan-

loicland, 2019-07-25:
Have you tried the latest commit? Otherwise reinstall Eigen.

shobhitjdpsr, 2019-07-25:
I tried the commit which was there 24 hours ago approx

@loicland

shobhitjdpsr, 2019-07-25:
@loicland 

SEE Eigen is already in newwst version, what should I try next?
[image]

shobhitjdpsr, 2019-07-25:
Nevermind, This was fixed by editing partition/ply_c/ply_c.cpp and changing

#include <Eigen/Dense>
#include <Eigen/Eigenvalues>

to

#include <eigen3/Eigen/Dense>
#include <eigen3/Eigen/Eigenvalues

Now, I am getting following issue:-
[image]

loicland, 2019-07-25:
Check whether your anaconda is using python3.6m, and if not adapt the cmake options (and dont forget to destroy all the cmakefiles).

shobhitjdpsr, 2019-07-25:
[image]

Thanks, it worked, I changed to 3.7m in cmake, because thats what I had in my cmake

Now  @loicland 

cmake .. -DPYTHON_LIBRARY=$CONDAENV/lib/libpython3.7m.so -DPYTHON_INCLUDE_DIR=$CONDAENV/include/python3.7m -DBOOST_INCLUDEDIR=$CONDAENV/include -DEIGEN3_INCLUDE_DIR=$CONDAENV/include/eigen3

I am getting error
[image]

shobhitjdpsr, 2019-07-25:
Just for your reference:-
[image]
[image]

shobhitjdpsr, 2019-07-25:
@loicland 

Now I will summarize the whole problem that I am facing currently:-

While compiling libcp, and running cmake (cmake .. -DPYTHON_LIBRARY=$/home/spartan/anaconda3/lib/libpython3.7m.so -DPYTHON_INCLUDE_DIR=$/home/spartan/anaconda3/include/python3.7m -DBOOST_INCLUDEDIR=$/usr/local/lib -DEIGEN3_INCLUDE_DIR=$/home/spartan/anaconda3/include/eigen3
) 
Note:- I have python3.7m in my anaconda3/lib and it worked for ply_c

Now , I am getting the following errors:-
[image]

ALso while running locate libboost python, I am getting following errors:-

[image]
[image]
[image]

shobhitjdpsr, 2019-07-25:
@loicland 

What would you recommend to currently do in order to run cmake for cut_pursuit ???

(I am on Ubuntu 16.04)

loicland, 2019-07-26:
you should remove the `$` before your anaconda path.

shobhitjdpsr, 2019-07-26:
[image]

@loicland 
Even after doing the same, I am still getting errors

shobhitjdpsr, 2019-07-26:
[image]

loicland, 2019-07-26:
Have you tried the symbolic link fix I added today to the readme?

Otherwise change in cut pursuit/src/cmakelist

   ``` find_package(Boost 1.67.0 COMPONENTS numpy${PYTHONVERSION} etc```
To
    ```find_package(Boost 1.67.0 COMPONENTS numpy```

shobhitjdpsr, 2019-07-26:
[image]

Do you mean just the line highlighted to be changed?

shobhitjdpsr, 2019-07-26:
@loicland

shobhitjdpsr, 2019-07-26:
[image]
[image]

@loicland 
I did try the new command in readme for symbolic link, but still I am getting and error. Should I change the cmakeLists also?

loicland, 2019-07-26:
Anywhere bumpy appears. But try the symbolic link first.

shobhitjdpsr, 2019-07-26:
@loicland 

Do you mean to remove cmakecachefiles?

ALso on trying 
 cmake .. -DPYTHON_LIBRARY=/home/spartan/Anaconda3/envs/my_env/lib/libpython3.7m.so -DPYTHON_INCLUDE_DIR=/home/spartan/Anaconda3/envs/my_env/include/python3.7m -DBOOST_INCLUDEDIR=/home/spartan/Anaconda3/include -DEIGEN3_INCLUDE_DIR=/home/spartan/Anaconda3/envs/my_env/eigen3

I get this:-
[image]

Also, is path of DBOOST_INCLUDEDIR correct?

shobhitjdpsr, 2019-07-26:
[image]

Finally, cmake runs with a warning.

But now make has some issues

shobhitjdpsr, 2019-07-26:
[image]

loicland, 2019-07-26:
[This thread](https://stackoverflow.com/questions/19810940/ubuntu-linking-boost-python-fatal-error-pyconfig-cannot-be-found) adapted to your python version.

shobhitjdpsr, 2019-07-26:
@loicland 

No again a warning has popped up.
Please have a look

[image]

shobhitjdpsr, 2019-07-26:
@loicland 

I tried various other  things to export path, but issue persists....

Please have a look 

[image]
[image]

shobhitjdpsr, 2019-07-26:
[image]

@loicland  : Have a look at this too

shobhitjdpsr, 2019-07-26:
@loicland 

While doing cmake, I was getting following cmake-warning.

Submitting for your reference:-
[image]

loicland, 2019-07-26:
Isn't your python version 3.7m? Try the export with this, and dont forget to erase the cmakefiles before.

shobhitjdpsr, 2019-07-26:
@loicland 

I  am using 3.7m.

Check this

[image]
[image]
[image]

shobhitjdpsr, 2019-07-26:
@loicland : What do you think might be causing the same?

shobhitjdpsr, 2019-07-26:
[image]
[image]
[image]

shobhitjdpsr, 2019-07-26:
@loicland 
I am using 3.7m in export, but the error persists

loicland, 2019-07-26:
Your path has an upper case to the first letter of Anaconda which is not the case in your drive.

shobhitjdpsr, 2019-07-26:
@loicland 
Thanks for your help!

[image]

Is it finally built successfully???

How can I check ?

loicland, 2019-07-26:
Yep, it's done. Choose a dataset, open its .md file and follow the instructions.

shobhitjdpsr, 2019-07-29:
[image]

@loicland  : 
In order to run the highlighted pretrained model, do we necessarily need CUDA?

I tried putting pretrained models in results/s3dis/pretrained/best    after extracting like this:_

[image]

Then, I put test.sh in Desktop/superpoint_graph like this:-
[image]

Then after running this on terminal I get this:-
[image]
[image]

@loicland 
Am I doing it the right way? What to do with the errors?
Also do we necessarily need CUDA installed? aSking so as its not mentioned in requirements

shobhitjdpsr, 2019-07-29:
@loicland

On running ./test.sh ,

I get something like this:-

**original error: libcublas.so.9.0: cannot open shared object file: No such file or directory**

Some more screenshot for your reference:-
[image]
[image]

loicland, 2019-07-30:
you can use the option `--cuda 0`

Also it seems that cupy is not found. Try to install it through conda:
```
 conda install -c anaconda cupy 
```

nicolas-chaulet, 2019-07-30:
You won't be able to install cupy if you don't have the nvidia drivers or no GPU, which is fine. With the flag above it should all work.

shobhitjdpsr, 2019-07-30:
@loicland  @nicolas-chaulet 
I tried installing using
 conda install -c anaconda cupy 
[image]

Then I updated test.sh, located here:-

[image]

I tried testing with test.sh containing all this:-
[image]
[image]

Bit I get same error:-
[image]
[image]

On doing pip freeze I get this:-
[image]

I tried removing the one I did try running through through conda's cupy but I get error when running test.sh:-
[image]
[image]

Also, when I tried running through only cupy-cuda90==6.2.0 installed, I was still getting errors.(same has been attached in my previous yesterday's message)

@loicland  @nicolas-chaulet 
What would you recommend now?

loicland, 2019-07-30:
what are the output of:
```nvcc --version```
and
```conda list```
in a terminal (in text form please, more readable than screenshot of terminals)

shobhitjdpsr, 2019-07-30:
On running nvcc --version, I get:-
The program 'nvcc' is currently not installed. You can install it by typing:
sudo apt install nvidia-cuda-toolkit

Also, conda list gives the following:-

ARNING: The conda.compat module is deprecated and will be removed in a future release.
# packages in environment at /home/spartan/anaconda3/envs/my_env:
#
# Name                    Version                   Build  Channel
_libgcc_mutex             0.1                        main  
blas                      1.0                         mkl    anaconda
boost                     1.67.0                   py37_4    anaconda
bzip2                     1.0.8                h7b6447c_0    anaconda
ca-certificates           2019.5.15                     0    anaconda
certifi                   2019.6.16                py37_0    anaconda
chardet                   3.0.4                    pypi_0    pypi
cudatoolkit               10.1.168                      0    anaconda
cudnn                     7.6.0                cuda10.1_0    anaconda
cupy                      6.0.0            py37hc0ce245_0    anaconda
eigen                     3.3.7                hfd86e86_0  
eigen3                    3.3.7                         0    omnia
fastrlock                 0.4                      pypi_0    pypi
future                    0.17.1                   pypi_0    pypi
h5py                      2.9.0                    pypi_0    pypi
icu                       58.2                 h211956c_0    anaconda
idna                      2.8                      pypi_0    pypi
intel-openmp              2019.4                      243    anaconda
joblib                    0.13.2                   pypi_0    pypi
libboost                  1.67.0               h46d08c1_4    anaconda
libedit                   3.1.20181209         hc058e9b_0  
libffi                    3.2.1                hd88cf55_4  
libgcc-ng                 9.1.0                hdf63c60_0  
libgfortran-ng            7.3.0                hdf63c60_0    anaconda
libiconv                  1.15                 h63c8f33_5  
libstdcxx-ng              9.1.0                hdf63c60_0  
mkl                       2019.4                      243    anaconda
mkl_fft                   1.0.12           py37ha843d7b_0    anaconda
mkl_random                1.0.2            py37hd81dba3_0    anaconda
nccl                      1.3.5                cuda10.0_0    anaconda
ncurses                   6.1                  he6710b0_1  
numpy                     1.16.4                   pypi_0    pypi
numpy-base                1.16.4           py37hde5b4d6_0    anaconda
openssl                   1.1.1                h7b6447c_0    anaconda
pillow                    6.1.0                    pypi_0    pypi
pip                       19.1.1                   py37_0  
plyfile                   0.7                      pypi_0    pypi
py-boost                  1.67.0           py37h04863e7_4    anaconda
pynvrtc                   9.2                      pypi_0    pypi
python                    3.7.3                h0371630_0  
python-igraph             0.7.1.post6              pypi_0    pypi
pyzmq                     18.0.2                   pypi_0    pypi
readline                  7.0                  h7b6447c_5  
requests                  2.22.0                   pypi_0    pypi
scikit-learn              0.21.2                   pypi_0    pypi
scipy                     1.3.0                    pypi_0    pypi
setuptools                41.0.1                   py37_0  
six                       1.12.0                   pypi_0    pypi
sklearn                   0.0                      pypi_0    pypi
sqlite                    3.29.0               h7b6447c_0  
tk                        8.6.8                hbc83047_0  
torch                     1.1.0                    pypi_0    pypi
torchfile                 0.1.0                    pypi_0    pypi
torchnet                  0.0.4                    pypi_0    pypi
torchvision               0.3.0                    pypi_0    pypi
tornado                   6.0.3                    pypi_0    pypi
tqdm                      4.32.2                   pypi_0    pypi
transforms3d              0.3.1                    pypi_0    pypi
urllib3                   1.25.3                   pypi_0    pypi
visdom                    0.1.8.8                  pypi_0    pypi
websocket-client          0.56.0                   pypi_0    pypi
wheel                     0.33.4                   py37_0  
xz                        5.2.4                h14c3975_4  
zlib                      1.2.11               h7b6447c_3

loicland, 2019-07-31:
Hi,

pull a fresh commit, it seems like your `cuda_kernels.py` is outdated. 

@nicolas-chaulet I see you added a `pass` when cupy fail to import, but doesn't it cause an error line 42 of `cuda_kernels.py` when the import fails?

shobhitjdpsr, 2019-07-31:
@loicland @nicolas-chaulet 

When trying to run test.sh containing:-
[image]

I get the following error:-
Traceback (most recent call last):
  File "learning/main.py", line 444, in <module>
    main()
  File "learning/main.py", line 159, in main
    train_dataset, test_dataset, valid_dataset, scaler = create_dataset(args)
  File "/home/spartan/Desktop/SJ/superpoint_graph/learning/s3dis_dataset.py", line 55, in get_datasets
    trainlist, testlist, validlist, scaler = spg.scaler01(trainlist, testlist, validlist=validlist)
  File "/home/spartan/Desktop/SJ/superpoint_graph/learning/spg.py", line 53, in scaler01
    edge_feats = np.concatenate([ trainlist[i][3] for i in range(len(trainlist)) ], 0)
ValueError: need at least one array to concatenate
[image]

Also, I tried searching in s3dis.py, it is searching .h5 files, but there are no .h5 files in stanford dataset, see:-
[image]

shobhitjdpsr, 2019-07-31:
what can b done to get rid of this???

loicland, 2019-08-01:
Your S3DIS_DIR folder should have a `/data` folder, a `/features`folder, a `superpoint_graphs` folder and a `/parsed` folder. All of these contains folder by area, and inside a .h5 file per room. Follow the instructions (partition and parsing) and you should get there.

shobhitjdpsr, 2019-08-01:
[image]

whuhxb, 2023-02-13:
> In ply_c while doing MAKE, I am getting following error:
>
> [ 50%] Building CXX object CMakeFiles/ply_c.dir/ply_c.cpp.o /home/spartan/Desktop/superpoint_graph/partition/ply_c/ply_c.cpp:4:23: fatal [...]

How to solve this compilation error? Thanks.