loicland/superpoint_graph #141: low accuracy in custom dataset.

EricWu123, 2019-07-29:
hi~
I trained the spg in my custom dataset, but the network can not  fit it and results in low training accuracy. I tried to increase the max-epoches to 2000, but it still not works. the last train accuracy is as follows:
```
-> Train accuracy: 64.07470288624788, 	Loss: 0.8975430130958557
Epoch 1999/2000 (results/isprs/trainval_best):
```
**what should I do to improve the train accuracy?**

loicland, 2019-07-29:
Hi,

some info which could help:
- nature of your dataset (acquisition modality, size)
- what classes are you using / are they unbalanced
- is the elevation measurement consistent from one cloud to the other (you can use the `write_scalar.py` to visualize it with CloudCompare)
- performance of a baseline algorithm (pointnet with sliding windows for example, or some simple random forrest on the geof (geometric features) + rgb).

EricWu123, 2019-07-29:
Thank you for reply.
I use the common point cloud dataset [ISPRS](http://www2.isprs.org/commissions/comm3/wg4/3d-semantic-labeling.html), including the classes as follows(the Powerline is ignored):
```
(label,class,point numbers)
0 Powerline  546
1 Low vegetation 180850 
2 Impervious surfaces  193723
3 Car 4614
4 Fence/Hedge  12070
5 Roof 152045
6 Facade  27250
7 Shrub  47605
8 Tree 135173
```
before I use the SPG, I have used random forest classifier with some simple geometric features(like planarity, linearity,scattering and so on). I got the 0.99 train accuracy but low test accuracy of 0.64.
And I use SPG to train in this dataset, I got 0.64 train accuracy and 0.58 test accuracy lower than methods before.
I think maybe the unsupervised partion algorithm is not suited for this, the partion result is as follows, it seems like very scattered.  how to evaluate the quality of partion?
[image]

loicland, 2019-07-29:
The partition doesn't seem too terrible, but you can try to use the supervized partition method to learn better geometric features. Or try lower and higher `--reg_strength` for the geometric partition.

Is the scan flat? I suspect that the elevation is not consistent. Visualize the elevation feature on your cloud to make sure that buildings are indeed popping out. You can use the simple plane model `--elevation_plane_model 1` to see if it is better as well.

Can you post an example of the prediction, and where it gets things wrong?