loicland/superpoint_graph #144: Why it's pred_full+1 instead of pred_full in the visulization

KangweiLiu, 2019-07-31:
Hello，dear
Why it used pred_full+1 instead of just pred_full while visualizing the final result , such as in "prediction2ply(ply_file + "_pred.ply", xyz, pred_full+1, n_labels,  args.dataset)" ?

loicland, 2019-07-31:
Hi,

0 is a reserved value in the ground truth for unannotated data, so the first class starts at 1. However, the network annotates all data and hence never predict 'unannotated'. consequently, the output starts at class 0, with a shift of 1 comapred to the ground truth.

I will add an addendum in the README, this questions comes back a lot!

KangweiLiu, 2019-08-01:
I got it， thanks！

> Hi,
>
> 0 is a reserved value in the ground truth for unannotated data, so the first class starts at 1. However, the network annotates all data and hence never predict 'unannotated'. consequently, the [...]

I got it， thanks！

closed 2019-08-14 by loicland