loicland/superpoint_graph #146: Online Segmentation of Point Cloud Stream from Camera

chpohl, 2019-08-08:
Hi, 

thank you for your great work. Is your SPG (maybe using the supervised partition) suitable for segmenting a live point cloud stream from an RGBD camera (i.e. Kinect or something similar). I guess the point clouds are several orders of magnitude smaller than the ones used here (e.g. max ~100k). Can you estimate how long your method would take (starting with an unprocessed pointcloud) for such an application?

Thanks in advance

loicland, 2019-08-08:
Hi,

the current computational bottlenecks are:
(1) computation of the knn graph
(2) computation of the minimal partition with cut pursuit
(3) computation of the voronoi adjacency graph.
(4) writing all the intermediary files on the hard-drive

the rest operates on superpoints and is very very fast.

For (1) and (3), this is mainly because we use the default sklearn built in function which are very slow. If speed is an issue, there exist faster alternatives (like ANN). If you have RGBD cameras you could even utilize the lattice structure for super fast adjacency graph computation.

For (2), it can process around 1m edge/s right now (some parameter tuning might be required inside cut pursuit), so about 200k points/s with a 5-n adjacency structure. A much faster version of cut pursuit have been released ([PCP](https://github.com/1a7r0ch3/parallel-cut-pursuit)), but we are still  missing the python wrappers for now. Should be done by Fall.

For (4), writing the intermediary results on the hard drive / ssd is not necessary for a stream. However, this is not coded yet. furthermore you would have to make the model run on streams as well, as it takes a few seconds just to launch the model as of now.

In short I think that 1s for 100k shall be attainable with minimal effort.  With more work (adjacency from grid, parallel cut pursuit, etc), this could go down realistically by about an order of magnitude depending on what kind of CPU youa re using.

Is this the kind of runtime you are looking for?

chpohl, 2019-08-12:
Hi,

thanks for your fast reply. I think 1 fps is the lower bound for something that is usable for my purposes. I will have a try at it. Thanks for your help.

closed 2019-08-12 by chpohl