loicland/superpoint_graph #147: SPG with Handcrafted Partition

CharlieLeee, 2019-08-13:
Thanks for providing such a great work! 
I would be very grateful if you could help me with this problem

When I ran the partition command for s3dis dataset:
python partition/partition.py --dataset s3dis --ROOT_PATH /home/charlie/LiDAR/superpoint_graph/S3DIS --voxel_width 0.03 --reg_strength 0.03

after a while I got the error:

Timer : 226.7 / 1398.2 / 1142.9 
20 / 23---> hallway_3
    creating the feature file...
        adding object 1 : clutter_2
        adding object 2 : door_1
        adding object 3 : clutter_1
        adding object 4 : floor_1
        adding object 5 : wall_3
        adding object 6 : door_2
        adding object 7 : wall_1
        adding object 8 : clutter_3
        adding object 9 : ceiling_2
        adding object 10 : ceiling_1
        adding object 11 : wall_2
        adding object 12 : door_3
=========================
======== pruning ========
=========================
Voxelization into 77 x 69 x 91 grid
Reduced from 175075 to 20435 points (11.67%)
97% done             
    computing the superpoint graph...
        minimal partition...
        computation of the SPG...
Timer : 227.0 / 1400.4 / 1144.5 
21 / 23---> hallway_2
    creating the feature file...
sys:1: DtypeWarning: Columns (1) have mixed types. Specify dtype option on import or set low_memory=False.
Traceback (most recent call last):
  File "partition/partition.py", line 122, in <module>
    xyz, rgb, labels, objects = read_s3dis_format(data_file)
  File "/home/charlie/LiDAR/superpoint_graph/partition/provider.py", line 173, in read_s3dis_format
    xyz = np.ascontiguousarray(room_ver[:, 0:3], dtype='float32')
  File "/home/charlie/anaconda3/lib/python3.7/site-packages/numpy/core/numeric.py", line 632, in ascontiguousarray
    return array(a, dtype, copy=False, order='C', ndmin=1)
ValueError: could not convert string to float: '-9.1\x160'

loicland, 2019-08-13:
Hi,

This is most likely due to an error in the s3dis street. Have you applied the fix?

```

```cp S3DIS_fix.diff $S3DIS_DIR/data; cd $S3DIS_DIR/data; git apply S3DIS_fix.diff; rm S3DIS_fix.diff; cd -

CharlieLeee, 2019-08-14:
> Hi,
> 
> This is most likely due to an error in the s3dis street. Have you applied the fix?
> 
> ```
> 
> ```cp S3DIS_fix.diff $S3DIS_DIR/data; cd $S3DIS_DIR/data; git apply S3DIS_fix.diff; rm S3DIS_fix.diff; cd -
> ```

Thanks for replying! 

It turns out that I ran the git apply diff command before this error happened but the dataset somehow did not change accordingly. After manually changing the dataset, it worked. Thanks for pointing it out!

closed 2019-08-14 by CharlieLeee

CharlieLeee, 2019-08-27:
> Hi CharlieLeee, I am experiencing the same issue. Could you tell me what
> did you modify in the dataset? Thank you!

Hi Yifan,

Thanks for reaching out, the way I solve the problem was by manually
applying the difference in the S3DIS dataset, i.e. changing those corrupted
characters in the dataset.

Good luck!
Regards,
Charlie

longmalongma, 2020-04-22:
> Thanks for replying!
>
> It turns out that I ran the git apply diff command before this error happened but the dataset somehow did not change accordingly. After manually changing the dataset, it [...]

Hi,how to change the dataset for solving the problem?looking for warding to your reply.

loicland, 2020-04-22:
type the following command line in a terminal in the spg folder:
```
cp S3DIS_fix.diff $S3DIS_DIR/data; cd $S3DIS_DIR/data; git apply S3DIS_fix.diff; rm S3DIS_fix.diff; cd -
```

longmalongma, 2020-04-22:
l

longmalongma, 2020-04-22:
> type the following command line in a terminal in the spg folder:
> 
> ```
> cp S3DIS_fix.diff $S3DIS_DIR/data; cd $S3DIS_DIR/data; git apply S3DIS_fix.diff; rm S3DIS_fix.diff; cd -
> ```

> type the following command line in a terminal in the spg folder:
> 
> ```
> cp S3DIS_fix.diff $S3DIS_DIR/data; cd $S3DIS_DIR/data; git apply S3DIS_fix.diff; rm S3DIS_fix.diff; cd -
> ```

> type the following command line in a terminal in the spg folder:
> 
> ```
> cp S3DIS_fix.diff $S3DIS_DIR/data; cd $S3DIS_DIR/data; git apply S3DIS_fix.diff; rm S3DIS_fix.diff; cd -
> ```
when I  type the apply D3DIS fix.diff,I meet this problem:
a$ git apply S3DIS_fix.diff;
warning: Area_3/hallway_2/hallway_2.txt has type 100755, expected 100644
error: patch failed: Area_3/hallway_2/hallway_2.txt:926334
error: Area_3/hallway_2/hallway_2.txt: patch does not apply
warning: Area_5/hallway_6/Annotations/ceiling_1.txt has type 100755, expected 100644
error: patch failed: Area_5/hallway_6/Annotations/ceiling_1.txt:180386
error: Area_5/hallway_6/Annotations/ceiling_1.txt: patch does not apply
error: Area_6/copyRoom_1/copy_Room_1.txt: No such file or directory
do you know why?

loicland, 2020-04-22:
Could it be that you are trying to apply the patch multiple times? Do you still have the crash in `partition.py`?

longmalongma, 2020-04-22:
> Could it be that you are trying to apply the patch multiple times? Do you still have the crash in `partition.py`?

yes,Ｉ can not repair s3dis daraset,when i run python partition.py, i meet this problem:
pandas.errors.ParserError: Error tokenizing data. C error: Expected 6 fields in line 926337, saw 7

but i do not know how to manｕａlly change s3dis dataset.

loicland, 2020-04-22:
> Sorry,i have solved this problem.thanks for your reply

Could you summarize what the problem ended up being for future users?

longmalongma, 2020-04-22:
> > Sorry,i have solved this problem.thanks for your reply
> 
> Could you summarize what the problem ended up being for future users?
For ease of operation，Ｍｙ file name of s3dis dataset is S3DIS_DIR.