loicland/superpoint_graph #152: boost_numpy" is   considered to be NOT FOUND

CharlieLeee, 2019-09-20:
while I was building ply_c on the server with python3.6 I got the error:

The C compiler identification is GNU 5.4.0
-- The CXX compiler identification is GNU 5.4.0
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Found PythonLibs: /home/undergraduate/miniconda2/envs/spg/lib/libpython3.6m.so (found version "3.6.9") 
-- Found PythonInterp: /home/undergraduate/miniconda2/envs/spg/bin/python3.6 (found version "3.6.9") 
-- Found NumPy: 1.16.5 (Required is at least version "1.5") 
Boost 1.65.0 found.
Found Boost components:
   graph
CMake Error at /usr/local/lib/cmake/Boost-1.71.0/BoostConfig.cmake:117 (find_package):
  Found package configuration file:

    /usr/local/lib/cmake/boost_numpy-1.71.0/boost_numpy-config.cmake

  but it set boost_numpy_FOUND to FALSE so package "boost_numpy" is
  considered to be NOT FOUND.  Reason given by package:

  No suitable build variant has been found.

  The following variants have been tried and rejected:

  * libboost_numpy27.a (2.7, Boost_PYTHON_VERSION=3.6)

  * libboost_numpy27.so.1.71.0 (2.7, Boost_PYTHON_VERSION=3.6)

Call Stack (most recent call first):
  /usr/local/lib/cmake/Boost-1.71.0/BoostConfig.cmake:182 (boost_find_component)
  /usr/share/cmake-3.5/Modules/FindBoost.cmake:245 (find_package)
  CMakeLists.txt:25 (find_package)

-- Configuring incomplete, errors occurred!
See also "/home/undergraduate/ChengkunLi/superpoint_graph/partition/ply_c/CMakeFiles/CMakeOutput.log".

Thanks for helping!

loicland, 2019-09-20:
Hi,

are you using a mac or a linux machien? Eitherway check out [this thread](https://github.com/loicland/superpoint_graph/issues/132) and let me know if you are still stuck.

loic

CharlieLeee, 2019-09-24:
I'm using a linux machine, specifically, ubuntu 14.04. After I tried to comment out include(FindNumPy), I still get the same error: 
The following variants have been tried and rejected:

libboost_numpy27.a (2.7, Boost_PYTHON_VERSION=3.6)

libboost_numpy27.so.1.71.0 (2.7, Boost_PYTHON_VERSION=3.6)

loicland, 2019-09-24:
Can you post the exact command line you enter to compute the cmake and the full error log in the terminal?

CharlieLeee, 2019-09-24:
Hi, thanks for answering! The command line I entered was 
`cmake . -DPYTHON_LIBRARY=/home/charlie/anaconda3/bin/anaconda/lib/libpython3.6m.so -DPYTHON_INCLUDE_DIR=/home/charlie/anaconda3/bin/anaconda/include/python3.6m -DBOOST_INCLUDEDIR=/home/charlie/anaconda3/bin/anaconda/include -DEIGEN3_INCLUDE_DIR=/home/charlie/anaconda3/bin/anaconda/include/eigen3`

and the log I got was 
The C compiler identification is GNU 5.4.0
-- The CXX compiler identification is GNU 5.4.0
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Found PythonLibs: /home/undergraduate/miniconda2/envs/spg/lib/libpython3.6m.so (found version "3.6.9")
-- Found PythonInterp: /home/undergraduate/miniconda2/envs/spg/bin/python3.6 (found version "3.6.9")
-- Found NumPy: 1.16.5 (Required is at least version "1.5")
Boost 1.65.0 found.
Found Boost components:
graph
CMake Error at /usr/local/lib/cmake/Boost-1.71.0/BoostConfig.cmake:117 (find_package):
Found package configuration file:

/usr/local/lib/cmake/boost_numpy-1.71.0/boost_numpy-config.cmake
but it set boost_numpy_FOUND to FALSE so package "boost_numpy" is
considered to be NOT FOUND. Reason given by package:

No suitable build variant has been found.

The following variants have been tried and rejected:

libboost_numpy27.a (2.7, Boost_PYTHON_VERSION=3.6)

libboost_numpy27.so.1.71.0 (2.7, Boost_PYTHON_VERSION=3.6)

Call Stack (most recent call first):
/usr/local/lib/cmake/Boost-1.71.0/BoostConfig.cmake:182 (boost_find_component)
/usr/share/cmake-3.5/Modules/FindBoost.cmake:245 (find_package)
CMakeLists.txt:25 (find_package)

-- Configuring incomplete, errors occurred!
See also "/home/undergraduate/ChengkunLi/superpoint_graph/partition/ply_c/CMakeFiles/CMakeOutput.log"\

Since I compiled the code perfectly on my own ubuntu machine and got the error while compiling on a public server of my school, so I guess it might because of the wrong version of boost, however, since I do not have the sudo permission on the server so I would like to know a possible way to solve the problem. Thanks again!

regards

closed 2019-10-06 by CharlieLeee

linlinge, 2021-07-27:
I face this problem when change the ubuntu 18.04 to ubuntu 20.04. When I change python3.7 to python 3.8, this problem solved.

iris0329, 2021-12-21:
Look at 

```python
libboost_numpy27.a (2.7, Boost_PYTHON_VERSION=3.6)

libboost_numpy27.so.1.71.0 (2.7, Boost_PYTHON_VERSION=3.6)
```

it's because the boost installed in your lab at path `/usr/local/lib/cmake/Boost-1.71.0` using python 2.7, but your conda environment is using python 3.6.

install another boost or python 2.7 conda  environment could solve this

rahidraghav, 2022-12-22:
i had a similar issue on Ubuntu 20.04 with python 3.10 .. installing  boost =1.74.0 solved the issue.

loicland, 2024-06-26:
Please use https://github.com/drprojects/superpoint_transformer
instead of this repo created in 2017. You can reproduce SPG there if you want to, and everything is up-to-date.