loicland/superpoint_graph #154: Train accuracy and loss are fine but Test Loss and accuracy are unreasonable??

ShichaoJin, 2019-10-08:
Step1:  I configure my environment using nvidia-docker by following README.md in your respository
Step2:  I run the Semantic3D examples by foloowing the Semantic3D.md 
## when I train on the whole publicly available data and test on the reduced test set, i.e., run the following code:
[image]

The results was shown bellow. The training accuracy and loss seem reasonable, which is increasing gradually, but the testing accuracy was quite small and the testing loss was much larger than the train loss. 

[image]

Thx for you help in advance,

loicland, 2019-10-08:
Hi,

this is because there is no ground truth on the test set of semantic3d.

closed 2019-10-17 by loicland

garryz94, 2020-05-09:
Hi @ShichaoJin , can you tell me how to compile libply_c and libcp in docker? Do i have to install anaconda in docker environment? I am kind of stuck here