loicland/superpoint_graph #156: visualization

longmalongma, 2019-10-11:
hello,i only have one rtx2080gpu,can i run your semantic segmentation on your three types of dataset?

loicland, 2019-10-11:
Hi. You'd probably be able.tomrun inference just fine. For training you might have to use smaller batches, and maybe change the norms to group norms.

Otherwise just use your CPU.

longmalongma, 2019-10-11:
ok,think you for reply,i am trying to run your codes. when do you provide some codes about scannet data?

longmalongma, 2019-10-11:
hi,i want to know if my one gpu memory is insufficient to load the dataset,two gpus do it?i have no idea that two gpus if can increase the memory?

loicland, 2019-10-17:
Hi,

I think yo should probably try it! I haven't tested this network on a multi-GPU setting personally.

longmalongma, 2019-10-18:
> Hi,
> 
> I think yo should probably try it! I haven't tested this network on a multi-GPU setting personally.
Thank your very much!

closed 2019-10-18 by longmalongma