loicland/superpoint_graph #157: pip install python-igraph is failed

longmalongma, 2019-10-12:
pip install python-igraph is failed,do you know why?
ERROR: Command errored out with exit status 1: /home/longma/anaconda2/envs/py3torch/bin/python -u -c 'import sys, setuptools, tokenize; sys.argv[0] = '"'"'/tmp/pip-install-qaitol0_/python-igraph/setup.py'"'"'; __file__='"'"'/tmp/pip-install-qaitol0_/python-igraph/setup.py'"'"';f=getattr(tokenize, '"'"'open'"'"', open)(__file__);code=f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' install --record /tmp/pip-record-kju6h6df/install-record.txt --single-version-externally-managed --compile Check the logs for full command output.

closed 2019-10-12 by longmalongma

yaping222, 2019-12-05:
Hi, I met the same issue. Could you demonstrate how to solve it? Thanks!

loicland, 2019-12-05:
is [this](https://stackoverflow.com/questions/28435418/failing-to-install-python-igraph) helpful?