loicland/superpoint_graph #159: Obtaining per-point predictions for original point cloud

mirceta, 2019-10-16:
Hello,

I've used your method with a custom dataset, posed basically as a binary classification problem. I am able to obtain the confusion matrix, but not an actual array of predictions for each point in the original pointcloud. Is this possible?

loicland, 2019-10-16:
Hi,

you can access the confusion matrix after evaluation with
```
print(confusion_matrix.confusion_matrix)
```

loicland, 2019-10-17:
You van use the function `visualize` as described in the readme.

mirceta, 2019-10-18:
Will check it, will certainly be better than only a confusion matrix. Just to clarify, there is then no way to actually export the per point predictions? Visualizing is nice, but I can't really compare it to the results of another algorithm and compare what each method is better at. For that, I'd need to import the predictions into my own tool for visualization.

loicland, 2019-10-18:
Hi,

you can access to the output of the network through `o_cpu`, but it actually gives you the result on the subsampled point clouds.

If you want the result on the original point cloud you can upsample the prediction 
with the function `interpolate_labels_batch`. This is done for the semantic3d dataset in the function `write_Semantic3d.py`. You might have to adapt the folder hierarchy to your own dataset.

mirceta, 2019-10-18:
Thank you, this is exactly what I needed. Will let you know how it goes.

loicland, 2019-10-18:
of note, I highly do not recommend to write the output in a csv file, as it is very slow compared to a h5 file for example. this is only done to comply with semantic3d's submission requirements.

mirceta, 2019-10-18:
Noted

mirceta, 2019-10-26:
Hi, I think I've gotten stuck a bit. There's 2 places where there's an `o_cpu` variable. It's in the training loop and the `eval_final`. In both cases `o_cpu` has only 64 elements. This seems very weird? I thought you'd get predictions for the downsampled point cloud. What am I missing?

mirceta, 2019-10-26:
Actually I see in the way the confusion matrix is computed, that these are actually all of the points but partitioned into 64 batches, and each point in the batch is assigned the same label. Is this an error? I'm a bit confused.