loicland/superpoint_graph #160: Semantic3D training error

CharlieLeee, 2019-10-17:
I got:
```
File "learning/main.py", line 444, in <module>
    main()
  File "learning/main.py", line 320, in main
    acc, loss, oacc, avg_iou = train()
  File "learning/main.py", line 194, in train
    outputs = model.ecc(embeddings)
  File "/home/undergraduate/miniconda3/envs/sp/lib/python3.7/site-packages/torch/nn/modules/module.py", line 547, in __call__
    result = self.forward(*input, **kwargs)
  File "/home/undergraduate/ChengkunLi/backup/superpoint_graph/learning/../learning/graphnet.py", line 97, in forward
    input = module(input)
  File "/home/undergraduate/miniconda3/envs/sp/lib/python3.7/site-packages/torch/nn/modules/module.py", line 547, in __call__
    result = self.forward(*input, **kwargs)
  File "/home/undergraduate/ChengkunLi/backup/superpoint_graph/learning/../learning/modules.py", line 54, in forward
    input = ecc.GraphConvFunction(nc, nc, idxn, idxe, degs, degs_gpu, self._edge_mem_limit)(hx, weights)
  File "/home/undergraduate/ChengkunLi/backup/superpoint_graph/learning/../learning/ecc/GraphConvModule.py", line 67, in forward
    cuda_kernels.conv_aggregate_fw(output.narrow(0,startd,numd), products.view(-1,self._out_channels), self._degs_gpu.narrow(0,startd,numd))
  File "/home/undergraduate/ChengkunLi/backup/superpoint_graph/learning/../learning/ecc/cuda_kernels.py", line 125, in conv_aggregate_fw
    function, stream = get_kernel_func('conv_aggregate_fw_kernel_v2', conv_aggregate_fw_kernel_v2(), get_dtype(src))
  File "/home/undergraduate/ChengkunLi/backup/superpoint_graph/learning/../learning/ecc/cuda_kernels.py", line 38, in get_kernel_func
    prog = Program(ksrc, kname+dtype+'.cu')        
NameError: name 'Program' is not defined
```
When running :
```
CUDA_VISIBLE_DEVICES=0 python learning/main.py --dataset sema3d --SEMA3D_PATH /home/undergraduate/ChengkunLi/Sem3 --db_test_name testred --db_train_name trainval --epochs 500 --lr_steps '[350, 400, 450]' --test_nth_epoch 100 --model_config 'gru_10,f_8' --ptn_nfeat_stn 11 --nworkers 2 --pc_attrib xyzrgbelpsv --odir "results/sema3d/trainval_best"
```

loicland, 2019-10-17:
Hi,
Check out learning/ecc/cuda_kernels.py line 38, there is a comment explaining how to fix this error.

CharlieLeee, 2019-10-17:
Thanks! That worked

closed 2019-10-17 by CharlieLeee

kxhit, 2019-12-27:
Hi! I got the same error and follow the comment try to fix this error. But I still get error.

`File "learning/main.py", line 444, in <module>
    main()
  File "learning/main.py", line 320, in main
    acc, loss, oacc, avg_iou = train()
  File "learning/main.py", line 194, in train
    outputs = model.ecc(embeddings)
  File "/media/work/anaconda3/envs/spg/lib/python3.6/site-packages/torch/nn/modules/module.py", line 489, in __call__
    result = self.forward(*input, **kwargs)
  File "/media/work/3D/superpoint_graph/learning/../learning/graphnet.py", line 97, in forward
    input = module(input)
  File "/media/work/anaconda3/envs/spg/lib/python3.6/site-packages/torch/nn/modules/module.py", line 489, in __call__
    result = self.forward(*input, **kwargs)
  File "/media/work/3D/superpoint_graph/learning/../learning/modules.py", line 54, in forward
    input = ecc.GraphConvFunction(nc, nc, idxn, idxe, degs, degs_gpu, self._edge_mem_limit)(hx, weights)
  File "/media/work/3D/superpoint_graph/learning/../learning/ecc/GraphConvModule.py", line 67, in forward
    cuda_kernels.conv_aggregate_fw(output.narrow(0,startd,numd), products.view(-1,self._out_channels), self._degs_gpu.narrow(0,startd,numd))
  File "/media/work/3D/superpoint_graph/learning/../learning/ecc/cuda_kernels.py", line 125, in conv_aggregate_fw
    function, stream = get_kernel_func('conv_aggregate_fw_kernel_v2', conv_aggregate_fw_kernel_v2(), get_dtype(src))
  File "/media/work/3D/superpoint_graph/learning/../learning/ecc/cuda_kernels.py", line 36, in get_kernel_func
    prog = Program(ksrc.encode('utf-8'), (kname+dtype+'.cu').encode('utf-8'))
NameError: name 'Program' is not defined
`
Could you help me?

loicland, 2019-12-27:
You must comment the line
```
prog = Program(ksrc, kname+dtype+'.cu') 
```
and uncomment the line
```
prog = Program(ksrc.encode('utf-8'), (kname+dtype+'.cu').encode('utf-8'))
```
or vice versa.

What is your version of pytorch?

kxhit, 2019-12-27:
Both of them are tried but nothing works. I'm using pytorch1.0.0 cuda 10.0 and python3.6.7.

loicland, 2019-12-27:
Could you post the error you get for each configuration? just the last part , from `File " in get_kernel_func prog = ...`

kxhit, 2019-12-27:
OK!
using:
`prog = Program(ksrc.encode('utf-8'), (kname+dtype+'.cu').encode('utf-8'))`
error:
`File "/media/work/3D/superpoint_graph/learning/../learning/ecc/cuda_kernels.py", line 36, in get_kernel_func
    prog = Program(ksrc.encode('utf-8'), (kname+dtype+'.cu').encode('utf-8'))
NameError: name 'Program' is not defined`

And another.

using:
`prog = Program(ksrc, kname+dtype+'.cu')`

error:
`File "/media/work/3D/superpoint_graph/learning/../learning/ecc/cuda_kernels.py", line 38, in get_kernel_func
    prog = Program(ksrc, kname+dtype+'.cu')
NameError: name 'Program' is not defined`

loicland, 2019-12-27:
Strange.

at the top of the function, can you add a print to see what is loaded:
`
try:
    import cupy.cuda
    print('"cupy loaded")
    from pynvrtc.compiler import Program
    print('"pynvrtc loaded")
except:
    print('"loading failed")
    pass

kxhit, 2019-12-27:
Thank you for your patience!
I find the inside error by following your suggestion, which is the cupy error. So I reinstall cupy with `pip install cupy-cuda100` (my cuda version is 10.0) and it works now!
Thank you again! Cheers!