loicland/superpoint_graph #161: supervized_partition.py: error: argument --ROOT_PATH: expected one argument

longmalongma, 2019-10-18:
when I run git apply S3DIS_fix.diff
I meet this problem:
error: Area_3/hallway_2/hallway_2.txt: No such file or directory
error: Area_5/hallway_6/Annotations/ceiling_1.txt: No such file or directory
error: Area_6/copyRoom_1/copy_Room_1.txt: No such file or directory

when I run 
for FOLD in 1 2 3 4 5 6; do \
    python ./supervized_partition/supervized_partition.py --ROOT_PATH $S3DIS_DIR  --cvfold $FOLD \
    --odir results_partition/s3dis/best --epochs 50 --reg_strength 0.1 --spatial_emb 0.2  \
    --global_feat eXYrgb --CP_cutoff 25; \
done
I meet this problem:
supervized_partition.py: error: argument --ROOT_PATH: expected one argument

longmalongma, 2019-10-18:
> What did you set `S3DIS_DIR` to? Try in a terminal:
> 
> ```
> echo $S3DIS_DIR
> ```

Thanks for your reply.I set it following your readme,Ａfter I use S3DIS_DIR instead of $S3DIS，it works.

closed 2019-10-18 by longmalongma

jing-zhao9, 2023-01-11:
@longmalongma 
How to set S3DIS DIR?

loicland, 2023-01-11:
This is the path where you put the S3DIS root folder.

jing-zhao9, 2023-01-11:
Do I need $export S3DIS DIR ='PATH'？