loicland/superpoint_graph #163: NameError: name 'n_labels' is not defined

longmalongma, 2019-10-21:
Hi,when I want to try to run:
python partition/visualize.py --dataset vkitti --ROOT_PATH VKITTI3D_DIR --res
_file 'results/vkitti/pretrained/cv1/predictions_test' --file_path '01/0001_00000' --output_type ifprs
I meet this problem:
writing the RGB file...
writing the features file...
writing the partition file...
writing the prediction file...
Traceback (most recent call last):
  File "partition/visualize.py", line 97, in <module>
    prediction2ply(ply_file + "_pred.ply", xyz, pred_full+1, n_labels,  args.dataset)
NameError: name 'n_labels' is not defined
can you help me ?looking forward to your reply.

longmalongma, 2019-10-21:
by the way, on vkitti datast, can not write GT.py.

loicland, 2019-11-14:
Hi,

sorry for the delay. 

It was indeed a missing line, it is fixed in the latest commit.

closed 2021-04-22 by loicland