loicland/superpoint_graph #164: How is the rgb information using?

garryzhi, 2019-10-22:
Hi loic, thanks for your great work! here is my question:
I can'i find where the rgb goes precisely in the code. All i can tell is that you read rgb from raw data and save it in the feature file(.h5). Do you re-extract it from features, or read it from dataloader? And the function load_superpoints() from learning/spg.py seems not to be called in the code and i am so confused

garryzhi, 2019-10-22:
it seems that rgb is used in the training and inference process of pointnet, but i can't find the data flow anyhow

loicland, 2019-10-22:
Hi,

once the SPG is computed with the `partition.py` function, the superpoints are computed and stored contiguously in a `.h5` file. For each superpoint, we have an array of size N_points x 14, as seen line 146 of `s3dis_dataset.py`:
```
 P = np.concatenate([xyz, rgb, e[:,np.newaxis], lpsv, xyzn, distance_to_center[:,None]], axis=1)
```
so the first 3 columns are the normalized position of the points , then colour, then  geometric features.

These .h5 files are read by the `load_superpoint` function in `spg.py`. This function is called line 150 of spg in the `loader` function. This function is called by the loader, for example line 171 of `main.py`. The refrence to loader is built in `train_dataset`, as described line 57 of `s3dis_dataset.py`.

garryzhi, 2019-10-23:
Thanks for your reply, it is really helpful.
It seems that rgb is used in the training and embedding process of PointNet. Is is also used in geometric partition and training process of ECC?

garryzhi, 2019-10-23:
And could you please make a precise definition of test accuracy and test oAcc as the training output shows:
-> Train accuracy: 87.13783277282879,   Loss: 0.3990295973195824,       Test accuracy: 78.40695166498361,       Test oAcc: 0.922440825767779,  Test avgIoU: 0.3548460575191651

-> Train accuracy: 87.36623845865645,   Loss: 0.3896398070038007,       Test accuracy: 73.60586453229607,       Test oAcc: 0.8592132397145127,         Test avgIoU: 0.40186568841530806

-> Train accuracy: 89.10910921193195,   Loss: 0.33808417842593247,      Test accuracy: 89.9864108414058,        Test oAcc: 0.9687821520353529,         Test avgIoU: 0.48897796711887237

loicland, 2019-10-23:
Hi,

the color is not used (yet) to learn the superedge filters.

OA = (TP + TN) / (TP + FP + FN + TN)
Acc = TP / (TP + FP)
IoU = TP / (TP + FP + FN)

garryzhi, 2019-10-23:
So oA means the proportion of correctly classified points?
It is easy to understand ACC for each class. But it seems that the Test Accuracy from training output is a measurement for the whole dataset. In this multi-class classification task, what does False Positive mean?

closed 2021-04-12 by loicland