loicland/superpoint_graph #165: problem on ablation stutdy

loklex, 2019-10-24:
Hi! Thanks for your great work!
I'm studying the performance of ECC. May I know the parameter --model_config, especially the number of nrepeats you use to do the ablation study of CRF-ECC in the picture below? 
[image]

loicland, 2019-10-25:
Hi,

this is the command line we used to produce the results in the ablation study (warning - untested in this branch):
```
for FOLD in 1 2 3 4 5 6; do \
CUDA_VISIBLE_DEVICES=0 python learning/main.py --dataset s3dis --S3DIS_PATH $S3DIR_DIR --cvfold $FOLD --epochs 350 --lr_steps '[275,320]' \
--test_nth_epoch 50 --model_config 'crf_10' --ptn_widths '[[64,64,128,128,256], [256,64,13]]' --ptn_nfeat_stn 14 --nworkers 2 --odir "results/s3dis/CRF−ECC/cv${FOLD}"; \
done
```

so 10 repeats.

loklex, 2019-10-28:
Thank you so much!

closed 2019-10-28 by loklex