loicland/superpoint_graph #167: CMake Warning:   Manually-specified variables were not used by the project:

zzusunquan, 2019-10-31:
While running `CONDAENV=/home/s206/anaconda3/envs/python36`
`cmake .. -DPYTHON_LIBRARY=$CONDAENV/lib/libpython3.6m.so 
-DPYTHON_INCLUDE_DIR=$CONDAENV/include/python3.6m
-DBOOST_INCLUDEDIR=$CONDAENV/include
 -DEIGEN3_INCLUDE_DIR=$CONDAENV/include/eigen3`
i  get a  warning:CMake Warning:
  Manually-specified variables were not used by the project:

    BOOST_INCLUDEDIR
    EIGEN3_INCLUDE_DIR

and  I  continue to run make , there is  no trouble 
and when i run  
 `python partition/partition.py --dataset sema3d --ROOT_PATH $SEMA3D_DIR --voxel_width 0.05 --reg_strength 0.8 --ver_batch 5000000` 

Traceback (most recent call last):
  File "partition/partition.py", line 15, in <module>
    import libcp
ImportError: ./partition/cut-pursuit/build/src/libcp.so: undefined symbol: _ZTIN5boost6python7objects21py_function_impl_baseE

ths same problem as     #61 
and  i try the  answer the problem is a linking problem with libboost. can you try @yxchng's solution in this thread (BVLC/caffe#5810)?

I look forward to hearing from you. I am very helpless now.

zzusunquan, 2019-10-31:
My python version is 3.6. The boost version is 1.5.3. I changed Boost 1.7.0 on CMakeLists.txt 20 22 25 lines to Boost 1.7.0 and changed 25 lines numpy${PYTHONVERSION} into numpy to make my cmake. Only Warning appears

zzusunquan, 2019-11-01:
Running the `locate libboost-python` system seems to be unresponsive
run `cmake --version`
output  :cmake version 3.16.0-rc2
run `locate anaconda3/lib/libpython`
output    /home/s206/anaconda3/lib/libpython3.7m.a
              /home/s206/anaconda3/lib/libpython3.7m.so
             /home/s206/anaconda3/lib/libpython3.7m.so.1
             /home/s206/anaconda3/lib/libpython3.7m.so.1.
run` conda list`
output  
# packages in environment at /home/s206/anaconda3/envs/python36:
#
# Name                    Version                   Build  Channel
_libgcc_mutex             0.1                        main  
asn1crypto                1.2.0                    py36_0  
blas                      1.0                         mkl  
boost                     1.70.0           py36h9de70de_1    conda-forge
boost-cpp                 1.70.0               h8e57a91_2    conda-forge
bzip2                     1.0.8                h7b6447c_0  
ca-certificates           2019.10.16                    0    anaconda
cairo                     1.16.0            hfb77d84_1002    conda-forge
certifi                   2019.9.11                py36_0    anaconda
cffi                      1.13.1           py36h2e261b9_0  
chardet                   3.0.4                 py36_1003  
cryptography              2.8              py36h1ba5d50_0  
cuda100                   1.0                           0    pytorch
cupy                      6.5.0                    pypi_0    pypi
eigen                     3.3.7                hfd86e86_0  
eigen3                    3.3.7                         0    omnia
fastrlock                 0.4                      pypi_0    pypi
fontconfig                2.13.1            h86ecdb6_1001    conda-forge
freetype                  2.9.1                h8a8886c_1  
future                    0.18.2                   pypi_0    pypi
gettext                   0.19.8.1          hc5be6a0_1002    conda-forge
glib                      2.58.3            h6f030ca_1002    conda-forge
gmp                       6.1.2             hf484d3e_1000    conda-forge
h5py                      2.10.0                   pypi_0    pypi
icu                       64.2                 he1b5a44_1    conda-forge
idna                      2.8                      py36_0  
igraph                    0.7.1             h9e3b1fc_1007    conda-forge
intel-openmp              2019.4                      243  
joblib                    0.14.0                   pypi_0    pypi
jpeg                      9b                   h024ee3a_2  
libedit                   3.1.20181209         hc058e9b_0  
libffi                    3.2.1                hd88cf55_4  
libgcc-ng                 9.1.0                hdf63c60_0  
libgfortran-ng            7.3.0                hdf63c60_0  
libiconv                  1.15              h516909a_1005    conda-forge
libpng                    1.6.37               hbc83047_0  
libsodium                 1.0.16               h1bed415_0  
libstdcxx-ng              9.1.0                hdf63c60_0  
libtiff                   4.0.10               h2733197_2  
libuuid                   2.32.1            h14c3975_1000    conda-forge
libxcb                    1.13              h14c3975_1002    conda-forge
libxml2                   2.9.10               hee79883_0    conda-forge
mkl                       2019.4                      243  
mkl-service               2.3.0            py36he904b0f_0  
mkl_fft                   1.0.14           py36ha843d7b_0  
mkl_random                1.1.0            py36hd6b4f25_0  
ncurses                   6.1                  he6710b0_1  
ninja                     1.9.0            py36hfd86e86_0  
numpy                     1.17.2           py36haad9e8e_0  
numpy-base                1.17.2           py36hde5b4d6_0  
olefile                   0.46                     py36_0  
openssl                   1.1.1                h7b6447c_0    anaconda
pcre                      8.43                 he1b5a44_0    conda-forge
pillow                    6.2.0            py36h34e0f95_0  
pip                       19.3.1                   py36_0  
pixman                    0.38.0            h516909a_1003    conda-forge
plyfile                   0.7.1                    pypi_0    pypi
[... 21 lines cut ...]
torchnet                  0.0.3                         0    srivasv
torchvision               0.2.1                      py_2    pytorch
tornado                   6.0.3            py36h7b6447c_0  
tqdm                      4.36.1                   pypi_0    pypi
transforms3d              0.3.1                    pypi_0    pypi
urllib3                   1.24.2                   py36_0  
visdom                    0.1.8.5                       0    srivasv
websocket-client          0.56.0                   py36_0  
wheel                     0.33.6                   py36_0  
xorg-kbproto              1.0.7             h14c3975_1002    conda-forge
xorg-libice               1.0.10               h516909a_0    conda-forge
xorg-libsm                1.2.3             h84519dc_1000    conda-forge
xorg-libx11               1.6.9                h516909a_0    conda-forge
xorg-libxau               1.0.9                h14c3975_0    conda-forge
xorg-libxdmcp             1.1.3                h516909a_0    conda-forge
xorg-libxext              1.3.4                h516909a_0    conda-forge
xorg-libxrender           0.9.10            h516909a_1002    conda-forge
xorg-renderproto          0.11.1            h14c3975_1002    conda-forge
xorg-xextproto            7.3.0             h14c3975_1002    conda-forge
xorg-xproto               7.0.31            h14c3975_1007    conda-forge
xz                        5.2.4                h14c3975_4  
yaml                      0.1.7                had09818_2  
zeromq                    4.3.1                he6710b0_3  
zlib                      1.2.11               h7b6447c_3  
zstd                      1.3.7                h0b5b093_0

loicland, 2019-11-01:
It seems like you haven't properly installed libboost.

Do:
```
sudo apt-get install libboost-all-dev
```
And make sure the link is updated in your path.

zzusunquan, 2019-11-01:
(python36) s206@s206-Precision-5820-Tower:~/Documents/sunquan/superpoint_graph-ssp-spg/partition/ply_c$ CONDAENV=/home/s206/anaconda3/envs/python36
(python36) s206@s206-Precision-5820-Tower:~/Documents/sunquan/superpoint_graph-ssp-spg/partition/ply_c$ cmake . -DPYTHON_LIBRARY=$CONDAENV/lib/libpython3.6m.so -DPYTHON_INCLUDE_DIR=$CONDAENV/include/python3.6m -DBOOST_INCLUDEDIR=$CONDAENV/include -DEIGEN3_INCLUDE_DIR=$CONDAENV/include/eigen3
-- Found PythonLibs: /home/s206/anaconda3/envs/python36/lib/libpython3.6m.so (found version "3.6.9") 
-- Found Boost 1.70.0 at /home/s206/anaconda3/envs/py37/lib/cmake/Boost-1.70.0
--   Requested configuration: QUIET REQUIRED COMPONENTS graph
-- Found boost_headers 1.70.0 at /home/s206/anaconda3/envs/py37/lib/cmake/boost_headers-1.70.0
-- Found boost_graph 1.70.0 at /home/s206/anaconda3/envs/py37/lib/cmake/boost_graph-1.70.0
--   libboost_graph.a
-- Adding boost_graph dependencies: regex;headers
-- Found boost_regex 1.70.0 at /home/s206/anaconda3/envs/py37/lib/cmake/boost_regex-1.70.0
--   libboost_regex.a
-- Adding boost_regex dependencies: headers
-- Found Boost: /home/s206/anaconda3/envs/py37/lib/cmake/Boost-1.70.0/BoostConfig.cmake (found suitable version "1.70.0", minimum required is "1.7.0") found components: graph 
-- Found Boost 1.70.0 at /home/s206/anaconda3/envs/py37/lib/cmake/Boost-1.70.0
--   Requested configuration: QUIET REQUIRED COMPONENTS numpy
-- Found boost_numpy 1.70.0 at /home/s206/anaconda3/envs/py37/lib/cmake/boost_numpy-1.70.0
--   libboost_numpy37.a
-- Adding boost_numpy dependencies: python;headers
-- Found boost_python 1.70.0 at /home/s206/anaconda3/envs/py37/lib/cmake/boost_python-1.70.0
--   libboost_python37.a
-- Adding boost_python dependencies: headers
-- Found Boost: /home/s206/anaconda3/envs/py37/lib/cmake/Boost-1.70.0/BoostConfig.cmake (found suitable version "1.70.0", minimum required is "1.7.0") found components: numpy 
Boost includes ARE /home/s206/anaconda3/envs/py37/include
Boost LIBRARIES ARE /home/s206/anaconda3/envs/py37/lib
PYTHON LIBRARIES ARE /home/s206/anaconda3/envs/python36/lib/libpython3.6m.so
-- Configuring done
-- Generating done
CMake Warning:
  Manually-specified variables were not used by the project:

    BOOST_INCLUDEDIR

-- Build files have been written to: /home/s206/Documents/sunquan/superpoint_graph-ssp-spg/partition/ply_c
(python36) s206@s206-Precision-5820-Tower:~/Documents/sunquan/superpoint_graph-ssp-spg/partition/ply_c$ make
Scanning dependencies of target ply_c
[ 50%] Building CXX object CMakeFiles/ply_c.dir/ply_c.cpp.o
In file included from /home/s206/Documents/sunquan/superpoint_graph-ssp-spg/partition/ply_c/ply_c.cpp:14:0:
/home/s206/Documents/sunquan/superpoint_graph-ssp-spg/partition/ply_c/connected_components.cpp: In function ‘void connected_components(uint32_t, uint32_t, const uint32_t*, const uint32_t*, const char*, std::vector<unsigned int>&, std::vector<std::vector<unsigned int> >&, uint32_t)’:
/home/s206/Documents/sunquan/superpoint_graph-ssp-spg/partition/ply_c/connected_components.cpp:64:38: warning: comparison between signed and unsigned integer expressions [-Wsign-compare]
    for (int i_ver_com = 0; i_ver_com < components_tmp[i_com].size(); i_ver_com++)
                                      ^
In file included from /home/s206/Documents/sunquan/superpoint_graph-ssp-spg/partition/ply_c/ply_c.cpp:14:0:
/home/s206/Documents/sunquan/superpoint_graph-ssp-spg/partition/ply_c/connected_components.cpp:75:48: warning: comparison between signed and unsigned integer expressions [-Wsign-compare]
      if (components_tmp[candidate_comp].size() > largest_neigh_comp_value)
                                                ^
In file included from /home/s206/Documents/sunquan/superpoint_graph-ssp-spg/partition/ply_c/ply_c.cpp:15:0:
/home/s206/Documents/sunquan/superpoint_graph-ssp-spg/partition/ply_c/random_subgraph.cpp: In function ‘void subgraph::random_subgraph(int, int, const uint32_t*, const uint32_t*, int, uint8_t*, uint8_t*)’:
/home/s206/Documents/sunquan/superpoint_graph-ssp-spg/partition/ply_c/random_subgraph.cpp:29:44: warning: comparison between signed and unsigned integer expressions [-Wsign-compare]
             for (uint32_t i_edg = 0; i_edg < n_edg; i_edg++)
                                            ^
/home/s206/Documents/sunquan/superpoint_graph-ssp-spg/partition/ply_c/random_subgraph.cpp:33:44: warning: comparison between signed and unsigned integer expressions [-Wsign-compare]
             for (uint32_t i_ver = 0; i_ver < n_ver; i_ver++)
                                            ^
/home/s206/Documents/sunquan/superpoint_graph-ssp-spg/partition/ply_c/random_subgraph.cpp:48:40: warning: comparison between signed and unsigned integer expressions [-Wsign-compare]
         for (uint32_t i_edg = 0; i_edg < n_edg; i_edg++)
                                        ^
/home/s206/Documents/sunquan/superpoint_graph-ssp-spg/partition/ply_c/ply_c.cpp: In function ‘PyObject* compute_geof(const boost::python::numpy::ndarray&, const boost::python::numpy::ndarray&, int)’:
/home/s206/Documents/sunquan/superpoint_graph-ssp-spg/partition/ply_c/ply_c.cpp:403:43: warning: comparison between signed and unsigned integer expressions [-Wsign-compare]
         for (std::size_t i_nei = 0; i_nei < k_nn; i_nei++)
                                           ^
[... cut ...]
zstd                      1.3.7                h0b5b093_0  
(python36) s206@s206-Precision-5820-Tower:~/Documents/sunquan/superpoint_graph-ssp-spg$ python partition/partition.py --dataset sema3d --ROOT_PATH $SEMA3D_DIR --voxel_width 0.05 --reg_strength 0.8 --ver_batch 5000000
Traceback (most recent call last):
  File "partition/partition.py", line 15, in <module>
    import libcp
ImportError: ./partition/cut-pursuit/build/src/libcp.so: undefined symbol: _ZTIN5boost6python7objects21py_function_impl_baseE

loicland, 2019-11-01:
Either libboost python is not installed, or it is not properly linked in your `LD_LIBRARY_PATH`

See [this issue](https://github.com/arvidn/libtorrent/issues/1743).

zzusunquan, 2019-11-02:
I really feel your reply, I posted the question you posted to me, I don't understand too much. I reconfigured a python environment to install some necessary packages. Can you give me some suggestions? I deleted the previous py3.6 environment. Run the following in the base environment
run locate anaconda3/lib/libpython
output
/home/s206/anaconda3/lib/libpython3.7m.a
/home/s206/anaconda3/lib/libpython3.7m.so
/home/s206/anaconda3/lib/libpython3.7m.so.1
/home/s206/anaconda3/lib/libpython3.7m.so.1.
Should I use python3.7 ，Previous boost I searched for download from anconda cloud. Do I need to uninstall it too?   
Is there any other package to note? I can't install the python-igraph package mentioned in the readme directly. I installed it in the anaconda cloud search. This will affect it.

zzusunquan, 2019-11-04:
Sorry, bother you, if my data is multi-dimensional, such as [x, y, z, R, G, B, Intensity, Verciticality], change your network input to 8D application on my data.
Can it be achieved?

loicland, 2019-11-04:
Hi,

Yes it can easily be done, you should adapt line 214 of ./learning/spg.py to your specific use case.

Also please post unrelated questions in new issues.

closed 2019-11-05 by zzusunquan