loicland/superpoint_graph #169: IndexError: too many indices for array

zzusunquan, 2019-11-05:
Hi  I am sorry to bother you.
while I run  python learning/sema3d_dataset.py --SEMA3D_PATH /home/s206/Documents/sunquan/superpoint_graph-ssp-spg/SEMA3D_DIR
Traceback (most recent call last):
  File "learning/sema3d_dataset.py", line 139, in <module>
    preprocess_pointclouds(args.SEMA3D_PATH)
  File "learning/sema3d_dataset.py", line 96, in preprocess_pointclouds
    for file in os.listdir(pathC):
FileNotFoundError: [Errno 2] No such file or directory: '/home/s206/Documents/sunquan/superpoint_graph-ssp-spg/SEMA3D_DIR/superpoint_graphs/train/'
Then  i change the sema3d_dataset.py  line 90  features_supervision to features 
 
 run  python learning/sema3d_dataset.py --SEMA3D_PATH 
/home/s206/Documents/sunquan/superpoint_graph-ssp-spg/SEMA3D_DIRdomfountain_station3.h5
Traceback (most recent call last):
  File "learning/sema3d_dataset.py", line 139, in <module>
    preprocess_pointclouds(args.SEMA3D_PATH)
  File "learning/sema3d_dataset.py", line 103, in preprocess_pointclouds
    hard_labels = np.argmax(labels[:,1:],1)
IndexError: too many indices for array

then I debug the sema3d_dataset.py  i found 

[image]
The labels array is empty and causes this error?
So I want to ask where to put the .labels file.
 SEMA3D_DIR/data? or SEMA3D_DIR?  or SEMA3D_DIR/data/train?
Or what other possible problems caused this error？

loicland, 2019-11-06:
Hi,

"The label files of the training files must be put in the same directory than the .txt files.", so SEMA3D_DIR/data/train.

you must relaunch the partition step however.

zzusunquan, 2019-11-06:
> Hi,
> 
> "The label files of the training files must be put in the same directory than the .txt files.", so SEMA3D_DIR/data/train.
> 
> you must relaunch the partition step however.

Thank you very much for your answer. I would like to ask if the labels file needs to decompressed. 
And I don’t know whether the label array is empty cause this problem .
When I use h5 explore to open the domfountain_station3.h5 there is nothing in the branch of labels.

loicland, 2019-11-06:
You download the labels in plain text from the semantic3d website.

For the train set, features/features_supervision must have a field `labels` of size N by 9. in the test/test_reduced it is just a simple scalar : 0.

zzusunquan, 2019-11-06:
Ok, thank you very much for your patience. Since I am late at night, I will go to the studio tomorrow and continue experiment. Thank you for giving me hope to continue. I wish you all the best

closed 2019-11-07 by zzusunquan

longmalongma, 2019-12-09:
> You download the labels in plain text from the semantic3d website.
> 
> For the train set, features/features_supervision must have a field `labels` of size N by 9. in the test/test_reduced it is just a simple scalar : 0.

> You download the labels in plain text from the semantic3d website.
> 
> For the train set, features/features_supervision must have a field `labels` of size N by 9. in the test/test_reduced it is just a simple scalar : 0.

Hello,i meet this problem:
 IndexError: too many indices for array
what is the field　＂ｌａｂｅｌｓ＂?how to ｏrganize "labes" files have download from website? Could you be more specific？looking forward your reply.

longmalongma, 2019-12-09:
> Thank you very much for your answer. I would like to ask if the labels file needs to decompressed.
> And I don’t know whether the label array is empty cause this problem .
> When I use h5 explore to [...]

Hello,have you solved this problem,where can download the labels in plain text from the semantic3d website?

zzusunquan, 2019-12-09:
> > > 嗨，
> > > “训练文件的标签文件必须与.txt文件放在同一目录中。”，所以SEMA3D_DIR / data / train。
> > > 但是，您必须重新启动分区步骤。
> > 
> > 
> > 非常感谢您的回答。我想问一下标签文件是否需要解压缩。
> > 而且我不知道标签数组是否为空会导致此问题。
> > 当我使用h5浏览打开domfountain_station3.h5时，标签的分支中没有任何内容。
> 
> 您好，您已经解决了这个问题，在哪里可以从语义3d网站下载纯文本标签？
[image]
This is  semantic website
http://www.semantic3d.net/view_dbase.php?chl=2

longmalongma, 2019-12-09:
> > > > 嗨，
> > > > “训练文件的标签文件必须与.txt文件放在同一目录中。”，所以SEMA3D_DIR / data / train。
> > > > 但是，您必须重新启动分区步骤。
> > > 
> > > 
> > > 非常感谢您的回答。我想问一下标签文件是否需要解压缩。
> > > 而且我不知道标签数组是否为空会导致此问题。
> > > 当我使用h5浏览打开domfountain_station3.h5时，标签的分支中没有任何内容。
> > 
> > 
> > 您好，您已经解决了这个问题，在哪里可以从语义3d网站下载纯文本标签？
> > [image]
> > This is  semantic website
> > http://www.semantic3d.net/view_dbase.php?chl=2

Hi,thank you for your reply. I have downloaded it. Which folder should I put it in?

zzusunquan, 2019-12-09:
> > > > > 嗨，
> > > > > “训练文件的标签文件必须与.txt文件放在同一目录中。”，所以SEMA3D_DIR /数据/火车。
> > > > > 但是，您必须重新启动分区步骤。
> > > > 
> > > > 
> > > > 非常感谢您的回答。我想问一下标签文件是否需要解压缩。
> > > > 而且我不知道标签上面是否为空会导致此问题。
> > > > 当我使用h5浏览打开domfountain_station3.h5时，标签的分支中没有任何内容。
> > > 
> > > 
> > > 您好，您已经解决了这个问题，在哪里可以从语义3d网站下载纯文本标签？？
> > > 这是语义网站http://www.semantic3d.net/view_dbase.php?chl=2
> > > [image]
> > > [](http://www.semantic3d.net/view_dbase.php?chl=2)
> 
> 嗨，谢谢你的答复。我已经下载了。我应该放入哪个文件夹？

"The label files of the training files must be put in the same directory than the .txt files.", so SEMA3D_DIR/data/train.

longmalongma, 2019-12-09:
> "The label files of the training files must be put in the same directory than the .txt files.", so SEMA3D_DIR/data/train.

> "The label files of the training files must be put in the same directory than the .txt files.", so SEMA3D_DIR/data/train.

I have put it as you said,but I still have this problem:
Traceback (most recent call last):
  File "learning/sema3d_dataset.py", line 139, in <module>
    preprocess_pointclouds(args.SEMA3D_PATH)
  File "learning/sema3d_dataset.py", line 103, in preprocess_pointclouds
    hard_labels = np.argmax(labels[:,1:],1)
IndexError: too many indices for array
do you know why?

longmalongma, 2019-12-09:
[image]

zzusunquan, 2019-12-09:
> > > > > > > 嗨，
> > > > > > > “训练文件的标签文件必须与.txt文件放在同一目录中。”，所以SEMA3D_DIR /数据/火车。
> > > > > > > 但是，您必须重新启动分区步骤。
> > > > > > 
> > > > > > 
> > > > > > 非常感谢您的回答。我想问一下标签文件是否需要解压缩。
> > > > > > 而且我不知道标签上面是否为空会导致此问题。
> > > > > > 当我使用H5浏览打开domfountain_station3.h5时，标签的分支中没有任何内容。
> > > > > 
> > > > > 
> > > > > 您好，您已经解决了这个问题，在哪里可以从语义3d网站下载纯文本标签？？
> > > > > 这是语义网站http://www.semantic3d.net/view_dbase.php?chl=2
> > > > > [image]
> > > 
> > > 
> > > 嗨，谢谢你的答复。我已经下载了。我应该放入该文件夹？
> > 
> > 
> > “培训文件的标签文件必须与.txt文件放在同一目录中。”，所以SEMA3D_DIR / data / train。
> 
> > > > > > > 嗨，
> > > > > > > “训练文件的标签文件必须与.txt文件放在同一目录中。”，所以SEMA3D_DIR /数据/火车。
> > > > > > > 但是，您必须重新启动分区步骤。
> > > > > > 
> > > > > > 
> > > > > > 非常感谢您的回答。我想问一下标签文件是否需要解压缩。
> > > > > > 而且我不知道标签上面是否为空会导致此问题。
> > > > > > 当我使用H5浏览打开domfountain_station3.h5时，标签的分支中没有任何内容。
> > > > > 
> > > > > 
> > > > > 您好，您已经解决了这个问题，在哪里可以从语义3d网站下载纯文本标签？？
> > > > > 这是语义网站http://www.semantic3d.net/view_dbase.php?chl=2
> > > > > [image]
> > > 
> > > 
> > > 嗨，谢谢你的答复。我已经下载了。我应该放入该文件夹？
> > 
> > 
> > “培训文件的标签文件必须与.txt文件放在同一目录中。”，所以SEMA3D_DIR / data / train。
> 
> 我已经按照您说的说了，但是我仍然有这个问题：
> 追溯（最近一次呼叫最近）：
> 在
> preprocess_pointclouds（args.SEMA3D_PATH）中的
> 文件“ learning / sema3d_dataset.py” ，第139行，文件“ learning / sema3d_dataset.py” ，第103行，在preprocess_pointclouds中
> hard_labels = np.argmax（labels [：，1：]，1）
> IndexError：数组索引过多，
> 您知道为什么吗？

you must relaunch the partition step , Because the labels data is written into the array during the partitioning process

longmalongma, 2019-12-09:
> you must relaunch the partition step , Because the labels data is written into the array during the partitioning process

I have uploaded my SEMA3D_DIR/data/train, is it correct?

zzusunquan, 2019-12-09:
> I have uploaded my SEMA3D_DIR/data/train, is it correct?

Yes

longmalongma, 2019-12-09:
> you must relaunch the partition step , Because the labels data is written into the array during the partitioning process

can you uploaded your SEMA3D_DIR/data/train directory？

zzusunquan, 2019-12-09:
> > > > > > > > > 嗨，
> > > > > > > > > “训练文件的标签文件必须与.txt文件放在同一目录中。”，所以SEMA3D_DIR /数据/火车。
> > > > > > > > > 但是，您必须重新启动分区步骤。
> > > > > > > > 
> > > > > > > > 
> > > > > > > > 非常感谢您的回答。我想问一下标签文件是否需要解压缩。
> > > > > > > > 而且我不知道标签上面是否为空会导致此问题。
> > > > > > > > 当我使用H5浏览打开domfountain_station3.h5时，标签的分支中没有任何内容。
> > > > > > > 
> > > > > > > 
> > > > > > > 您好，您已经解决了这个问题，在哪里可以从语义3d网站下载纯文本标签？？
> > > > > > > 这是语义网站http://www.semantic3d.net/view_dbase.php?chl=2
> > > > > > > [image]
> > > > > 
> > > > > 
> > > > > 嗨，谢谢你的答复。我已经下载了。我应该放入该文件夹？
> > > > 
> > > > 
> > > > “培训文件的标签文件必须与.txt文件放在同一目录中。”，所以SEMA3D_DIR /数据/火车。
> > > 
> > > 
> > > > > > > > > 嗨，
> > > > > > > > > “训练文件的标签文件必须与.txt文件放在同一目录中。”，所以SEMA3D_DIR /数据/火车。
> > > > > > > > > 但是，您必须重新启动分区步骤。
> > > > > > > > 
> > > > > > > > 
> > > > > > > > 非常感谢您的回答。我想问一下标签文件是否需要解压缩。
> > > > > > > > 而且我不知道标签上面是否为空会导致此问题。
> > > > > > > > 当我使用H5浏览打开domfountain_station3.h5时，标签的分支中没有任何内容。
> > > > > > > 
> > > > > > > 
> > > > > > > 您好，您已经解决了这个问题，在哪里可以从语义3d网站下载纯文本标签？？
> > > > > > > 这是语义网站http://www.semantic3d.net/view_dbase.php?chl=2
> > > > > > > [image]
> > > > > 
> > > > > 
> > > > > 嗨，谢谢你的答复。我已经下载了。我应该放入该文件夹？
> > > > 
> > > > 
> > > > “培训文件的标签文件必须与.txt文件放在同一目录中。”，所以SEMA3D_DIR /数据/火车。
> > > 
> > > 
> > > 我已经按照您说的说了，但是我仍然有这个问题：
> > > 重新（最近一次呼叫最近）：
> > > 在
> > > preprocess_pointclouds（args.SEMA3D_PATH）中的
> > > 文件“ learning / sema3d_dataset.py”，第139行，文件“ learning / sema3d_dataset.py” ，，第103行，在preprocess_pointclouds中
> > > hard_labels = np.argmax（标签[：，1：]，1）
> > > IndexError：排序索引过多，
> > > 您知道为什么吗？
> > 
> > 
> > 您必须重新启动分区步骤，因为标签数据是在分区过程中写入数组的
> 
> 您可以上传SEMA3D_DIR / data / train目录吗？

[image]

longmalongma, 2019-12-09:
> [image]

Thanks your help,I will try to run it again.

longmalongma, 2019-12-09:
> [image]

Hello,I have relunched partition.py,then i run python learning/sema3d_dataset.py --SEMA3D_PATH SEMA3D_DIR,i still meet this problem:
 python learning/sema3d_dataset.py --SEMA3D_PATH SEMA3D_DIR
bildstein_station1.h5
bildstein_station3.h5
bildstein_station5.h5
domfountain_station1.h5
domfountain_station2.h5
domfountain_station3.h5
sg27_station1.h5
sg27_station2.h5
sg27_station4.h5
sg27_station5.h5
sg27_station9.h5
sg28_station4.h5
station1_xyz.h5
Traceback (most recent call last):
  File "learning/sema3d_dataset.py", line 139, in <module>
    preprocess_pointclouds(args.SEMA3D_PATH)
  File "learning/sema3d_dataset.py", line 103, in preprocess_pointclouds
    hard_labels = np.argmax(labels[:,1:],1)
IndexError: too many indices for array
can you help me  again?

loicland, 2019-12-09:
Hi.

`station1_xyz.h5` is not a correct filename.

Did you modify the `untermeadenbrunnen_station1_xyz.txt` file name somehow? If not redownload it and unzip it again.

longmalongma, 2019-12-09:
> Hi.
> 
> `station1_xyz.h5` is not a correct filename.
> 
> Did you modify the `untermeadenbrunnen_station1_xyz.txt` file name somehow? If not redownload it and unzip it again.

Thanks for you reply,I didn't modify any files,ｄo you mean redownload untermeadenbrunnen_station1_xyz.txt,and relunch partion.py and sema3d_dataset.py?

loicland, 2019-12-09:
Check in the features and superpoint graph folders if there is indeed a untermeadenbrunnen_station1.h5 file in each. If there is a station1_xyz.h5 it means something went wrong in the name parsing.

longmalongma, 2019-12-09:
> Check in the features and superpoint graph folders if there is indeed a untermeadenbrunnen_station1.h5 file in each. If there is a station1_xyz.h5 it means something went wrong in the name parsing.

[image]
I find the neugasse_station1_xyz_intensity_rgb.labels and station1_xyz_intensity_rgb.txt do match,need i modify station1_xyz_intensity_rgb.txt to neugasse_station1_xyz_intensity_rgb.txt?

kxhit, 2019-12-27:
I modify "station1_xyz.txt" to "neugasse_station1_xyz_intensity_rgb.txt" and everything goes well.