loicland/superpoint_graph #170: /learning/../learning/metrics.py:70: RuntimeWarning: invalid value encountered in true_divide   return sum(values) / class_seen

zzusunquan, 2019-11-07:
Hi while I run  CUDA_VISIBLE_DEVICES=0 python learning/main.py --dataset sema3d --SEMA3D_PATH $SEMA3D_DIR --db_test_name testred --db_train_name trainval \
--epochs 500 --lr_steps '[350, 400, 450]' --test_nth_epoch 100 --model_config 'gru_10,f_8' --ptn_nfeat_stn 11 \
--nworkers 2 --pc_attrib xyzrgbelpsv --odir "results/sema3d/trainval_best"
The following error was reported after the end of the train epoch 500
Epoch 499/500 (results/sema3d/trainval_best):
100%|██████████████████████████████| 7/7 [00:03<00:00,  1.77it/s]
-> Train Loss: 0.4289   Train accuracy: 86.38%
100%|██████████████████████████████| 4/4 [00:19<00:00,  4.79s/it]
-> Test Loss: 12.1779  Test accuracy: 1.20%  Test oAcc: 0.00%  Test avgIoU: nan%
100%|██████████████████████████████| 4/4 [00:19<00:00,  4.83s/it]
100%|██████████████████████████████| 4/4 [00:19<00:00,  4.78s/it]
100%|██████████████████████████████| 4/4 [00:19<00:00,  4.79s/it]
100%|██████████████████████████████| 4/4 [00:19<00:00,  4.83s/it]
100%|██████████████████████████████| 4/4 [00:19<00:00,  4.79s/it]
100%|██████████████████████████████| 4/4 [00:19<00:00,  4.80s/it]
100%|██████████████████████████████| 4/4 [00:19<00:00,  4.82s/it]
100%|██████████████████████████████| 4/4 [00:19<00:00,  4.82s/it]
100%|██████████████████████████████| 4/4 [00:19<00:00,  4.80s/it]
100%|██████████████████████████████| 4/4 [00:19<00:00,  4.82s/it]
/home/s206/Documents/sunquan/superpoint_graph-ssp-spg/learning/../learning/metrics.py:70: RuntimeWarning: invalid value encountered in true_divide
  return sum(values) / class_seen
-> Multisample 10: Test accuracy: 1.1612523886520676, 	Test oAcc: 0.0, 	Test avgIoU: nan, 	Test mAcc: 0.0
My test_full folder is empty, there is no data in it, but I commented on test_full in part of the code. I don't know why this error occurred, and the training accuracy rate is not so high.

zzusunquan, 2019-11-14:
I still can't solve this problem, can you give me some tips?

loicland, 2019-11-14:
Hi,

the accuracy is nonsensical because there is no ground truth in test_reduced (you launched it on test_red not test_full by the way). So you can ignore these values. if you want to see how well it performs, submit your .labels file on the semantic3d server.

zzusunquan, 2019-11-20:
Hi
while I run "We validated our configuration on a custom split of 11 and 4 clouds" this part 
 ```
CUDA_VISIBLE_DEVICES=0 python learning/main.py --dataset sema3d --SEMA3D_PATH $SEMA3D_DIR --epochs 450 --lr_steps '[350, 400]' --test_nth_epoch 100 \
--model_config 'gru_10,f_8' --pc_attrib xyzrgbelpsv --ptn_nfeat_stn 11 --nworkers 2 --odir "results/sema3d/best"
```
I meet the same problem as the train on the whole publicly available data and test on the reduced test set 。

> /home/s206/Documents/sunquan/superpoint_graph-ssp-spg/learning/../learning/metrics.py:70: RuntimeWarning: invalid value encountered in true_divide
>   return sum(values) / class_seen
> -> Multisample 10: Test accuracy: 0, 	Test oAcc: 0.0, 	Test avgIoU: nan, 	Test mAcc: 0.0

I would like to ask this 11 cloud test 4 cloud training why there is no test accuracy rate ，there is ground truth in 11 cloud and 4 cloud .

closed 2019-11-26 by zzusunquan