loicland/superpoint_graph #171: make: *** No targets specified and no makefile found.  Stop.

hurricane1997, 2019-11-08:
[image]
I have run the cmake file code successfully but when running "make". It could not find the makefile.

loicland, 2019-11-08:
Is a makefile built in `/partition/ply_c`?

hurricane1997, 2019-11-09:
[image]
I do not find the makefile in /partition/ply_c.
I set CONDAENV=E:/Anaconda

loicland, 2019-11-13:
Hi,

it seems like the cmake worked. What is your version of cmake? It needs 3.7 I think to compile.

I have never personnaly used Cmake on windows so I can only help you so much.

closed 2021-04-22 by loicland