loicland/superpoint_graph #173: The obtained label file has one less line than the real scene.

zzusunquan, 2019-11-20:
Hi 
while  i run 
`python partition/write_Semantic3d.py --SEMA3D_PATH $SEMA3D_DIR --odir "results/sema3d/trainval_best" --db_test_name testred`
The obtained label file has one less line than the real scene.
for example :
[image]
[image]
Predicted MarketplaceFeldkirch_Station4.label file is 10538632 lines ,but the real scene it is 10538633   . the same problem happend in the othen .label file ,just less one line .

loicland, 2019-11-21:
Hi,

sorry for delay. It is indeed a bug, it should be fixed now. Let me know if the latest commit fixes your issue.

zzusunquan, 2019-11-22:
Hi
I try your latest commit ,but a bug has appeared.
Traceback (most recent call last):
 ```
 File "partition/write_Semantic3d.py", line 77, in <module>
    labels_ups = interpolate_labels_batch(data_file, xyz, labels_full, args.ver_batch)
  File "/home/s206/Documents/sunquan/superpoint_graph-ssp-spg/partition/provider.py", line 620, in interpolate_labels_batch
    print("read lines %d to %d" % (i_rows, i_rows + ver_batch))
TypeError: unsupported operand type(s) for +: 'NoneType' and 'int'
```

zzusunquan, 2019-11-22:
> Hi,
> 
> It is indeed a bug, it should be fixed now. Let me know if the latest commit fixes your issue.
I  comment line 620,in interpolate_labels_batch  print("read lines %d to %d" % (i_rows, i_rows + ver_batch)) to solv the typeError .so that the code can tun .
But the labels less one line not solved .
[image]
[image]
If I had a problem with partitioning or some other step .

zzusunquan, 2019-11-25:
Hi！
I am sorry to bother you again, I tried your latest submission, and the result is still one less line. I want to ask if everything is normal after you modify the code.

loicland, 2019-11-25:
Hi,

Sorry for delay I am currently travelling and can't check the code right now. I will get to it tomorrow afternoon.

zzusunquan, 2019-11-25:
> 你好
> 
> 抱歉，我目前正在旅行，暂时无法检查代码。我明天下午去。
Thank you very much for your reply, I wish you a pleasant trip, I will try it myself, I will inform you in time, but my code level is very limited, I will try hard.

zzusunquan, 2019-11-25:
> 你好
> 
> 抱歉，我目前正在旅行，暂时无法检查代码。我明天下午去。
> 
> 同时，如果您想寻找自己，则该错误肯定是由于我将numpy.genfromtext替换为panda函数。
[image]
Thank you for your help, I have successfully reproduced the results, I wish you a pleasant journey.

closed 2019-11-25 by zzusunquan

loicland, 2019-11-25:
Great, could you post your fix please?

zzusunquan, 2019-11-25:
partition/write_Semantic3d.py   74 lines chang it to 
`    labels_ups = interpolate_labels_batch(data_file, xyz, labels_full, args.ver_batch) + 1
`
partition/provider.py        607 to 650 lines
`
def interpolate_labels_batch(data_file, xyz, labels, ver_batch):
    """interpolate the labels of the pruned cloud to the full cloud"""
    if len(labels.shape) > 1 and labels.shape[1] > 1:
        labels = np.argmax(labels, axis = 1)
    # i_rows = None
    i_rows = 0
    labels_f = np.zeros((0, ), dtype='uint8')
    #---the clouds can potentially be too big to parse directly---
    #---they are cut in batches in the order they are stored---
    nn = NearestNeighbors(n_neighbors=1, algorithm='kd_tree').fit(xyz)
    while True:
        try:
            if ver_batch>0:
                print("read lines %d to %d" % (i_rows,i_rows + ver_batch))
                # if i_rows is None:
                #     print("read lines %d to %d" % (0, ver_batch))
                # else:
                #     print("read lines %d to %d" % (i_rows, i_rows + ver_batch))                      
                vertices = np.genfromtxt(data_file
                         , delimiter=' ', max_rows=ver_batch
                       , skip_header=i_rows)
                # vertices = pd.read_csv(data_file
                #          , sep=' ', nrows=ver_batch
                #          , header=i_rows).values
            else:
                # vertices = np.genfromtxt(data_file
                #         , delimiter=' ')
                # vertices = pd.read_csv(data_file
                #          , delimiter=' ').values
                break
        except (StopIteration, pd.errors.ParserError):
            #end of file
            break
        if len(vertices)==0:
            break
        xyz_full = np.array(vertices[:, 0:3], dtype='float32')
        del vertices
        distances, neighbor = nn.kneighbors(xyz_full)
        del distances
        labels_f = np.hstack((labels_f, labels[neighbor].flatten()))
        i_rows = i_rows +ver_batch
        # if i_rows is None:
        #     i_rows = ver_batch
        # else:
        #     i_rows = i_rows + ver_batch
    return labels_f
`