loicland/superpoint_graph #175: Imbalanced Classes

yaoyao1900, 2019-11-22:
Hi loic, thanks for your great work! 

I trained the spg in my custom dataset, the dataset has  5 different classes without RGB and it is similar to semantic3d  dataset. Training and test worked well. 
But my dataset has a problem of classes imbalanced. I have tried this methode but it didn't work. 
`       class_sample_count = [1000, 1000, 50, 1000, 1000] 
        weights = 1 / torch.Tensor(class_sample_count)
        weights = weights.double()
        sampler = torch.utils.data.sampler.WeightedRandomSampler(weights, 
                                                                 num_samples = 2000,
                                                                 replacement = True)
        
        loader = torch.utils.data.DataLoader(train_dataset,
                                             batch_size = args.batch_size,
                                             collate_fn=spg.eccpc_collate,
                                             num_workers=args.nworkers,
                                             shuffle=False, 
                                             sampler = sampler,
                                             drop_last=True)`
The error
`Epoch 0/1000 (results/Vaihingen/train_test):
  0%|                                   | 0/1000 [00:00<?, ?it/s]Traceback (most recent call last):
  File "main_ClassImbalanced.py", line 515, in <module>
    main()
  File "main_ClassImbalanced.py", line 391, in main
    acc, loss, oacc, avg_iou = train()
  File "main_ClassImbalanced.py", line 214, in train
    for bidx, (targets, GIs, clouds_data) in enumerate(loader):
  File "/home/wang/anaconda3/envs/torch1.0.1/lib/python3.6/site-packages/tqdm/_tqdm.py", line 1017, in __iter__
    for obj in iterable:
  File "/home/wang/anaconda3/envs/torch1.0.1/lib/python3.6/site-packages/torch/utils/data/dataloader.py", line 615, in __next__
    batch = self.collate_fn([self.dataset[i] for i in indices])
  File "/home/wang/anaconda3/envs/torch1.0.1/lib/python3.6/site-packages/torch/utils/data/dataloader.py", line 615, in <listcomp>
    batch = self.collate_fn([self.dataset[i] for i in indices])
  File "/home/wang/anaconda3/envs/torch1.0.1/lib/python3.6/site-packages/torchnet/dataset/listdataset.py", line 49, in __getitem__
    super(ListDataset, self).__getitem__(idx)
  File "/home/wang/anaconda3/envs/torch1.0.1/lib/python3.6/site-packages/torchnet/dataset/dataset.py", line 14, in __getitem__
    raise IndexError("CustomRange index out of range")
IndexError: CustomRange index out of range
`
Can you give me some suggestions? Thank you for your help!!

loicland, 2019-11-22:
Hi,

i would use class weights instead of importance sampling. It is already implemented, see line 74-82 of `/learning/s3dis_dataset.py` for example. Use `--loss_weights proportional` for example.

yaoyao1900, 2019-11-26:
Thank you very much! That worked.

closed 2019-11-26 by yaoyao1900