loicland/superpoint_graph #177: error in cmake

typhoonlee, 2019-12-05:
Hello ，i encountered a problem during compilation.
Instead of using conda to install boost, I used source code compilation, and the version is boost_1.69.0。
The obtained library is libboost_python27, I don't know if the library in your CMakeList.txt file is libboost_python36 or libboost_python27
I hope you can reply to me, and I wish you a successful research

loicland, 2019-12-05:
I have never tried with python 2.7 but I don't recommend that you do. It should be libboost_python36, try to manually specify the version if need be.

typhoonlee, 2019-12-06:
Thank you very much. It works.
But there was an error again during make

typhoonlee, 2019-12-06:
fatal error: numpy/ndarrayobject.h: No such file or directory
How should I do

loicland, 2019-12-06:
Please check issue #13, #36 and #132. tell me if none of the fixes works for you.

typhoonlee, 2019-12-06:
Thank you very much, it has been fixed, after executing make, the following display：
[100%] Linking CXX shared library libply_c.so
[100%] Built target ply_c
Did this compile successfully?

closed 2021-04-12 by loicland