loicland/superpoint_graph #181: Questions about the partitioning process

zzusunquan, 2019-12-09:
Hi Loicland !
    I have two quenstions about partition:
    1) During partition,The compiler calls the cmake libply_c library and libcp library, finally generated h5 files in the primary key [' distances'] [' geof'] [' labels'] [' RGB'] [' source '] [' target '] [' xyz '], the [' distances'] [' geof] [' sources'] [' target '] represent what meaning, and [' geof] which geometric feature four columns of data are respectively.
   2) Can I compute the geometry by cloudcompare myself, and instead of partitioning, write the calculated geometry to an h5 file as a data source to generate the superpoint？

loicland, 2019-12-13:
[' sources'] -> origin vertex for all superedge
[' target ']  -> destination vertex for all superedge
[' distances'] -> euclidian length of edges
[' geof]] -> geometric features (linarity, planarity, scattering, verticality)

You can compute the geof yourself (although I dont think that our version of verticality is standard), but I am not sure why you would want to. Our implementation is quite efficient/

zzusunquan, 2019-12-16:
> ['source']->所有超边的原始顶点
> ['target']->所有超边的目标顶点
> ['distances]->边缘
> 的欧几里得长度['geof]]->几何特征（线性，平面度，散射，垂直度）
> 
> 您可以自己计算geof（尽管我不认为我们的垂直度版本是标准的），但是我不确定为什么要这么做。我们的实施效率很高/
Because I want to use the multi-scale features, and it is not only a vertical degree.
I carried out the operation:  The dataset is semantic3d
1) Remove the points labeled 0
2) Point cloud subsampling
3) Calculate multi-scale ( 0.05,0.1,0.25,0.5,1  Spherical radii of five scales) geometric features (Roughness ,Gaussian Curvature,Ominvariance,Linearty,Verciticality)    After calculating the features, there are 25 columns of geometric features and 32 columns of data with the original RGB intensity.
4) I'm going to replace the nan value of the feature that I didn't calculate with minus 1
5) Extract the label data separately
So now I have 32 columns of training data and a label data. May I ask how to embed it into your network for partitioning and the convolution after that? Would you please give me some Suggestions

loicland, 2019-12-27:
I am not sure why you would remove the points with labels 0. These are most points in semantic3d. You can interpolate the labels using the `partition/provider/interpolate_labels` function

I am confused what your objective here is. Is it to partition or do the semantic segmentation? For the later, the emebdding/convolution process are detailed in the SPG paper.

closed 2020-03-17 by zzusunquan