loicland/superpoint_graph #182: ValueError: results/sema3d/trainval_best/prediction_testred.h5 does not exist and is needed to output the result ply

longmalongma, 2019-12-10:
Hello,i run your code accoring to your readme,but when i want to visualize it ,i meet this problem:
python partition/visualize.py --dataset sema3d --ROOT_PATH SEMA3D_DIR --res_f
ile 'results/sema3d/trainval_best/prediction_testred' --file_path 'test_reduced/MarketplaceFeldkirch_Station4' --output_type ifprs
Traceback (most recent call last):
  File "partition/visualize.py", line 70, in <module>
    raise ValueError("%s does not exist and is needed to output the result ply" % res_file) 
ValueError: results/sema3d/trainval_best/prediction_testred.h5 does not exist and is needed to output the result ply
can you help me ?

loicland, 2019-12-10:
Did you run the inference to completion? Where is the model located? The results file should be in the same fodler. Right now it is looking in `results/sema3d/trainval_best/`, you should adapt this option.

longmalongma, 2019-12-10:
> Did you run the inference to completion? Where is the model located? The results file should be in the same fodler. Right now it is looking in `results/sema3d/trainval_best/`, you should adapt this option.

closed 2019-12-10 by longmalongma

longmalongma, 2019-12-10:
> Did you run the inference to completion? Where is the model located? The results file should be in the same fodler. Right now it is looking in `results/sema3d/trainval_best/`, you should adapt this option.

> Did you run the inference to completion? Where is the model located? The results file should be in the same fodler. Right now it is looking in `results/sema3d/trainval_best/`, you should adapt this option.

python partition/visualize.py --dataset sema3d --ROOT_PATH SEMA3D_DIR --res_f
ile 'results/sema3d/trainval_best/predictions_testred' --file_path 'test_reduced/MarketplaceFeldkirch_Station4' --output_type e
writing the error file...
Traceback (most recent call last):
  File "partition/visualize.py", line 101, in <module>
    error2ply(ply_file + "_err.ply", xyz, rgb, labels, pred_full+1)
  File "/media/longma/daima/superpoint_graph/partition/provider.py", line 77, in error2ply
    if len(labels.shape) > 1 and labels.shape[1] > 1:
AttributeError: 'list' object has no attribute 'shape'
can you help me ?