loicland/superpoint_graph #184: AttributeError: 'list' object has no attribute 'shape'

longmalongma, 2019-12-12:
python partition/visualize.py --dataset sema3d --ROOT_PATH SEMA3D_DIR --res_f
ile 'results/sema3d/trainval_best/predictions_testred' --file_path 'test_reduced/StGallenCathedral_station6' --output_type ge
when i want to visualize the ground truth and error label, i meet this problem:

Traceback (most recent call last):
  File "partition/visualize.py", line 85, in <module>
    prediction2ply(ply_file + "_GT.ply", xyz, labels, n_labels, args.dataset)
  File "/media/longma/daima/superpoint_graph/partition/provider.py", line 59, in prediction2ply
    if len(prediction.shape) > 1 and prediction.shape[1] > 1:
AttributeError: 'list' object has no attribute 'shape'

what's the matter?Besides, i use meshlab to can't open the generated rgb.py,

loicland, 2019-12-12:
There is no ground truth available for the test set, you can only submit your results on the semantic3d servers.

longmalongma, 2019-12-12:
> There is no ground truth available for the test set, you can only submit your results on the semantic3d servers.

thank you for your reply, why don't have ground truth available?how implement the qualitative comparison if don't have ground truth?

loicland, 2019-12-12:
You can submit your labeling on the semantic3d website, and you will receive an  email with the results.

If you want to do some more in depth analysis you can use our train-val 11/4 split of the training set.

longmalongma, 2019-12-12:
> You can submit your labeling on the semantic3d website, and you will receive an email with the results.
> 
> If you want to do some more in depth analysis you can use our train-val 11/4 split of the training set.

longmalongma, 2019-12-12:
> You can submit your labeling on the semantic3d website, and you will receive an email with the results.
> 
> If you want to do some more in depth analysis you can use our train-val 11/4 split of the training set.

how to evaluate the mIoU of model on sema3d dataset?

closed 2019-12-18 by longmalongma