loicland/superpoint_graph #186: When setting the validation set be the same with the training set avg_iou != avg_iou_val

CharlieLeee, 2019-12-16:
I set the training dataset the same with the validation dataset in order to find out the best model that fits the training data, however, when monitoring the training process, I found out the validation ious are quite different from training ious, is that normal?

loicland, 2019-12-16:
It's normal that they should be a little different (the method is stochastic), but not very different.

- is the difference always in the same direction?
- which dataset are you using, a custom one?
- check in debug mode the clouds[1] values in the training loop as well ad embeddings outputs by the pointnet inside the evaluation loop.
- set `--test_multisamp_n` to 1 and see if the problem persists

Let me know!

CharlieLeee, 2019-12-17:
At certain epoch, they are in the opposite direction
[image]

The dataset was split from the semantic3d dataset and I used the command:
```
--use_val_set 
```

I'll try to set `--test_multisamp_n` to 1 and see the result

Thanks

loicland, 2019-12-17:
I think I have missunderstood what you did. What is your training and valdiation set comprised of?

The figure you posted seems ok to me. The method is doubly stochastic: random subgraph are extracted during training, so it is seen differently in train and eval mode; the superpoints are randomly sampled.

CharlieLeee, 2019-12-17:
Thanks for the explanation! the training set and validation set are exactly the same, so I expected the same value for val iou and train iou, so as you've said, these two processes were done on different downsamples of the original data? So I should expect a different result?

loicland, 2019-12-19:
Different downsampling for superpoints, but also at training time only a subgraph of the full spg is extracted to allows for larger batch and graph augmentation. As you can see however, even if train and eval are not exactly identical they evolves similarily.

CharlieLeee, 2019-12-19:
@loicland Right, thanks! also, another question, if I want to use validation set and save the best model during the validation, should I alter the argument "-save_nth_epoch" to be larger than the epoch I'm training, are there any other parameters/code I should change if I want to use a validation set?

CharlieLeee, 2019-12-27:
@loicland Hi loicland, would you please tell me how to train the model using validation set, and test with the best performance model on the validation set, it seems that in the 'main.py', the program will automatically save the model at the last epoch as the final model?

loicland, 2019-12-27:
Hi,

use `--use_val_set 1` to save the epoch that performs best on the validation set.

CharlieLeee, 2019-12-27:
@loicland should I also add --use_val_set 1 when performing the testing?

loicland, 2019-12-27:
No need because you set epochs to -1 for testing. Note that when doping the inference on semantic3d  the metrics will be nonsensical because the ground truth is withheld. you should use the function `write_semantic3d` in `\partition` to write the labels files and upload them on the semantic3d servers.

CharlieLeee, 2019-12-27:
[image]

@loicland After training with validation and testing at the end of the training, the model's epoch changed to 500 which is the number of training epoch, does that mean the training save only the model from the last epoch?
Also, as for the "epoch" in the scores*.json file
[image]
does this imply the total number of epoch or the best epoch?

closed 2020-03-14 by CharlieLeee