loicland/superpoint_graph #187: Retrain your trained ｗｅｉｇｈｔ

longmalongma, 2019-12-18:
Hello,I have a confusion,why we need to run python ./learning/main.py to your trained weights?Is it train again and can change the parameter of your trained weights?

loicland, 2019-12-19:
you add the `--resume RESUME` and `--epochs -1` to directly use our trained weight, as explained in the corresponding READMEs.

longmalongma, 2019-12-19:
> you add the --resume RESUME and --epochs -1 to directly use our trained weight, as explained in the corresponding READMEs.

yes, I have used your trained wetght, but I need to run  ./learning/main.py accroding to your DEADMEs. Whether executing this command will change your trained weight?What does this command do?

loicland, 2019-12-19:
It won't change the weight. because epochs = -1, it will go straight to inference on the test set and skip any training.

closed 2021-04-12 by loicland