loicland/superpoint_graph #188: What's "We validated our configuration on a custom split of 11 and 4 clouds" meaning？

GSusan, 2019-12-27:
hi, sorry to disturb.
During my training on Semantic3D, I couldn't clearly understand the difference between "To train on the whole publicly available data and test on the reduced test set, run"  and "We validated our configuration on a custom split of 11 and 4 clouds" in Readme. 
1. what's datasets ‘test' for when training?
2. when training with train and testred, I set --db_train_name trainval, is that mean training at the same time validating with val; for the same reason, setting --db_train_name train means training withou validating?
thanks

loicland, 2019-12-27:
We tuned the hyperparameters by training on 11 clouds (train) and sdelecting the best performing configuration on a custom validation set (val), see `get_datasets` in `learning/sema3d_dataset.py`. We then trained the best configuration on the entire train set (trainval = train + val).

For semantic3d the test labels are withheld so it is the only way to tune hyperparameters.

GSusan, 2019-12-31:
Thanks for your reply.
But what maks me confused is that, when I'm training with trainval, test dataset has no labels. Though predictions_testred(labels) are withheld, the Test accuracy is just about 1.32, Test oAcc=0.0, and so on.
Thanks again for your patience.

closed 2021-04-22 by loicland