loicland/superpoint_graph #192: Problem at execution time (undefined symbol)

sfoucher, 2020-01-25:
Hello,

I followed your instructions but I have this error when I execute (Ubuntu 16.0, python 3.6):

`$ python partition/partition.py --dataset s3dis --ROOT_PATH $S3DIS_DIR --voxel_width 0.03 --reg_strength 0.03

Traceback (most recent call last):
  File "partition/partition.py", line 16, in <module>
    import libply_c
ImportError: ./partition/ply_c/libply_c.so: undefined symbol: _ZTIN5boost6python7objects21py_function_impl_baseE  `

Here are the libraries used by cmake:
`Boost LIBRARIES ARE /mnt/d/DEV/linux/mic3/envs/SPG/lib/libboost_graph.so

PYTHON_LIBRARIES  ARE /mnt/d/DEV/linux/mic3/envs/SPG/lib/libpython3.6m.so`

Do I need to add libboost_graph.so in /usr/lib?

Thank you!

loicland, 2020-01-27:
Hi,

please check #167  and #61 . Let me know if you're problem remains.

sfoucher, 2020-02-01:
Thank you for your quick response, it seems to work now. Below are the steps starting from a clean installation of Ubuntu 16.0 under WSL:
* Step 1: update your Linux and install Miniconda
```   
sudo apt update && sudo apt upgrade
sudo apt intall g++
sudo apt install g++
sudo apt install cmake
bash Miniconda3-latest-Linux-x86_64.sh # from https://repo.continuum.io/miniconda/
exit
```
* Step 2: configure the conda env
```
conda create -n SPG python=3.6
conda activate SPG
sudo apt-get install build-essential g++ python-dev autotools-dev libicu-dev build-essential libbz2-dev libboost-all-dev
pip install git+https://github.com/pytorch/tnt.git@master --user
pip install future --user
sudo apt-get install xml2
sudo apt install zlib
sudo apt install zlib1g-dev
sudo apt-get install libxml2-dev
pip install python-igraph --user
pip install tqdm transforms3d pynvrtc fastrlock h5py sklearn plyfile scipy --user
conda install -c anaconda boost; conda install -c omnia eigen3; conda install eigen; conda install -c r libiconv
conda install pandas
```
* Step 3: compile the libraries
```    
   cd superpoint_graph/
   cd partition/
   cd ply_c/
   which python
   export CONDAENV=/mnt/c/DEV/linux/mc3/envs/SPG
   rm CMakeCache.txt
   cmake . -DPYTHON_LIBRARY=$CONDAENV/lib/libpython3.6m.so -DPYTHON_INCLUDE_DIR=$CONDAENV/include/python3.6m -DBOOST_INCLUDEDIR=$CONDAENV/include -DEIGEN3_INCLUDE_DIR=$CONDAENV/include/eigen3
   make
   python -c 'import libply_c'
```

closed 2020-02-01 by sfoucher

loicland, 2020-02-03:
thanks that is useful!