loicland/superpoint_graph #196: Compile the libply_c and libcp libraries on windows

fclairec, 2020-02-06:
Hi,
Thanks a lot for sharing your nice work!
I am having troubbles getting the Cmake to compile properly. Do you have experience in how to set up the command on windows? 

[image]

I updated Boost to the newest version and still there is the same message.
I am not familiar to the dependancies of Cmake so I am not sure if I am simply missing a little thing. 

I would greatly appreciate your help, thank you !

loicland, 2020-02-07:
Hi,

I have never tried personally to compile the library on windows myself.

See (this guide](https://www.boost.org/doc/libs/1_54_0/libs/python/doc/building.html) if it helps.

fclairec, 2020-02-17:
Tank you for the tip

closed 2020-02-17 by fclairec