loicland/superpoint_graph #199: Gated graph neural network

medif, 2020-02-25:
Hi, 
i read the paper many times bu  i still don't get why did you choose The gated graph neural  network.
 Can i have further explanation

loicland, 2020-02-26:
Hi,

so you mean the GRU, Gated Recurrent Unit? Is your question why a GRU specifically (as opposed to an LSTM for example), or why use a recurrent neural network at all?

medif, 2020-02-26:
yeah exactly and please i have an other question how pointnet could make data augmentation on the small partition (the superpoint) how can we learn information on small part (like a small part in a buildingfor exemple). I used your algrithm for a non rgb data and the partition result dont look very well. so i  wonder how the data augmentation can be done on such parts.
Thank you in advance

loicland, 2020-02-27:
The RNN is used to update the hidden states as messages a re propagated from one superpoint to another. This is the standard Graph Convolution approach.

We use GRUs as they are efficient, and we don't need to have two memories (long term and short terms). In the ablation study we try an LSTM, and it works similarly, despite an increase i parameter number.

 > how pointnet could make data augmentation on the small partition (the superpoint) how can we learn information on small part (like a small part in a building for exemple)

unsure I dunderstand. Superpoints that are too small have their state set to 0 instead of using a cloud embedder. Their hidden state is hence 100% determined by context.

> partition result dont look very well
Are you sure your classes are separable purely with geometric features?
This could be an issue with the hyperparameterization as well. Post a screenshot of the partition here and I'll see what I can say about it.
Have you tried a supervized partition approach otherwise?

closed 2021-04-12 by loicland