loicland/superpoint_graph #205: Visualization of ground truth missing labels

renegades12, 2020-04-20:
@loicland Hi, I ran the model on Semantic3D dataset, and after creating the superpoint .h5 file I ran the visualization command:
```bash
python partition/visualize.py --dataset sema3d --ROOT_PATH $s3d_path  --file_path 'train/sg27_station9' --output_type g
```
to visualize the gt point cloud, however, the Point cloud I got looks like this:

[image]
Is it right to get this subsampled one? or how to fix it? thanks!

renegades12, 2020-04-20:
@loicland 
But when I ran this:
```bash
python partition/visualize.py --dataset sema3d --ROOT_PATH $s3d_path --file_path 'train/sg27_station9' --output_type g
```

I got the **"seems right"** gt file:

[image]

loicland, 2020-04-22:
The ground truth of semantic 3D is sparse, and only a heavily subsampled subset of points are annotated. Depending on the resolution and the voxelisation it can be more or less apparent. It also varies from cloud to cloud throughout the dataset.

If you really want to have a 'full' ground truth, you can use the interpolating functions `interpolate_labels`.

renegades12, 2020-04-22:
@loicland thanks! that makes sense. in terms of using `interpolate_labels`, I tried to simply add an additional argument `--upsample 1` and then output the gt file, but the ply file looks the same

loicland, 2020-04-22:
I haven't coded the ground truth upsampling yet.
It would look like something like that in visualize.py:
```
if gt_out and bool(args.upsample):
    if args.dataset=='s3dis':
        #TODO
    elif args.dataset=='sema3d':
        data_file  = data_folder + file_name + ".txt"
        xyz_up, rgb_up = read_semantic3d_format(data_file, 0, '', 0, args.ver_batch)
    del rgb_up
    gt_up = interpolate_labels(xyz_up, xyz, label, args.ver_batch)
    print("writing the upsampled ground truth file...")
    prediction2ply(ply_file + "_gt_up.ply", xyz_up, gt_up +1, n_labels, args.dataset)
```

renegades12, 2020-04-24:
@loicland Hi, thanks for the code, I tried this out:

first, I set `ver_batch = 4`

Then after a long time of processing like this:
```
...
processing lines 44990624 to 44990628
processing lines 44990628 to 44990632
processing lines 44990632 to 44990636
processing lines 44990636 to 44990640
writing the upsampled ground truth file... 
```
the file I got `domfountain_station1_urban_trained_gt_up.ply` contains only 4 points, do you think there might be something wrong with the `ver_batch` size? or something else went wrong?

renegades12, 2020-04-24:
@loicland Hi, thanks for the code, I tried this out:

first, I set `ver_batch = 4`

Then after a long time of processing like this:
```
...
processing lines 44990624 to 44990628
processing lines 44990628 to 44990632
processing lines 44990632 to 44990636
processing lines 44990636 to 44990640
writing the upsampled ground truth file... 
```
the file I got `domfountain_station1_gt_up.ply` contains only 4 points, do you think there might be something wrong with the `ver_batch` size? or something else went wrong?

loicland, 2020-04-27:
Hi,

ver_batch is the number of lines to be read simultaneously. It should be around 5.000.000 rather than 4!

In case this doesn't fix your issue, could printthe following sizes:
```
print(xyz_up.shape)
print(xyz.shape)
print(label.shape)
```

renegades12, 2020-04-28:
@loicland Hi, thanks for the advice! I changed the size to 5,000,000 and print out the size
[image]

However, the output still looks wired:

[image]

So, I debugged the code and finally find out the cause:

Since there's no `label` in the file `visualize.py` so instead I use the `labels` from `geof, xyz, rgb, graph_nn, labels = read_features(fea_file)`, but the `labels` file looks like this:

[image]
Thus in the function `interpolate_labels`, labels will always be 0 
[image]
[image]
But as I look into the `labels` read from feature file, I was confused that there's a `10` in the array (index 211), but I expect the max of sem3d as 8
[image]

Do these look fine? or do you suggest which part did I made a mistake?
Looking forward to your suggestions!

closed 2023-06-16 by loicland