loicland/superpoint_graph #209: about custom_set

LeanderWayne, 2020-06-03:
hi, I tried to run the code on Oakland 3-D Point Cloud Dataset, it is without rgb. I have runed partition.py and custom_dataset.py successfully. But I got error when I tried to run learning/main.py
the error shows like this:
RuntimeError: Given groups=1, weight of size 64 8 1, expected input[143, 5, 128] to have 8 channels, but got 5 channels instead

I have used _--pc_attrib xyzelpsv_, the whole command are:
_CUDA_VISIBLE_DEVICES=0 python learning/main.py --dataset custom_dataset --CUSTOM_SET_PATH /home/leon/oakland --db_test_name test --db_train_name trainval \
--epochs 500 --lr_steps '[350, 400, 450]' --test_nth_epoch 100 --model_config 'gru_10,f_5' --ptn_nfeat_stn 8 \
--nworkers 2 --pc_attrib xyzelpsv --odir "results/oakland/trainval_best"_

I am not sure what I did wrong. Can you help me? @loicland

loicland, 2020-06-05:
Hi,

you are telling the STN to expect 8 values per point:

change `--ptn_nfeat_stn 8` to `--ptn_nfeat_stn 5`

LeanderWayne, 2020-06-07:
Hi, i met another problem, in learning/main.py function train(), `confusion_matrix.count_predicted_batch(tvec_cpu, np.argmax(o_cpu,1))`
the dataset has 5 classes, but i printed out the np.argmax(o_cpu,1) and there is 6 and 7 in the array, which cause IndexError because the confusion_matrix is 5*5.

zzusunquan, 2020-07-17:
> 嗨，我尝试在Oakland 3-D点云数据集上运行代码，因为它没有rgb。我已经成功运行partition.py和custom_dataset.py。但是，当我尝试运行learning / main.py时
> 出现错误，该错误显示如下：
> RuntimeError：给定groups = 1，权重为64 8 1，预期输入[ 143，5，128 ]具有8个通道，但是改为5个频道
> 
> 我用过_--pc_attrib xyzelpsv_，整个命令是：
> _CUDA_VISIBLE_DEVICES = 0 python learning / main.py --dataset custom_dataset --CUSTOM_SET_PATH / home / leon / oakland --db_test_name test --db_train_name trainval --epochs 500 --lr_steps '[350、400、450]'--test_nth_epoch 100 --model_config'gru_10，f_5'--ptn_nfeat_stn 8 --nworkers 2 --pc_attrib xyzelpsv --odir“结果/奥克兰/ trainval_best”_
> 
> 我不确定自己做错了什么。你能帮助我吗？@loicland

Nín hǎo qǐngwèn nín jiějuéle nín de wèntí ma wǒ yù dàole hé nín yīyàng de cuòwù
26/5000
Hello, did you solve your problem? I encountered the same error as you

closed 2021-04-12 by loicland