loicland/superpoint_graph #210: Can you expose OA and mIoU for each category on the Vkitti dataset

longmalongma, 2020-06-03:
Hi,thanks for your reply, Can you expose OA and mIoU for each category on the Vkitti dataset?looking forwarding to your reply.

loicland, 2020-06-05:
`python learning/evaluate.py --dataset vkitti --odir results/vkitti/best --cvfold 123456` should give you a breakdown of OA and IoU by classes.

longmalongma, 2020-06-13:
> `python learning/evaluate.py --dataset vkitti --odir results/vkitti/best --cvfold 123456` should give you a breakdown of OA and IoU by classes.
Thanks.

longmalongma, 2020-06-13:
> `python learning/evaluate.py --dataset vkitti --odir results/vkitti/best --cvfold 123456` should give you a breakdown of OA and IoU by classes.

Have you recorded　IoU  of each category before?Can you just give me mIoU of each category of your model on the vKITTI dataset right here? I want to quote your paper in my paper and compare it with your algorithm for mIoU of each category.

closed 2023-06-16 by loicland