loicland/superpoint_graph #212: some bugs happen with boost::graph while trying to compile in Win10,visual studio 2015,

zhairuoming, 2020-06-09:
I have install boost1.69 by conda ,and then tried to compile the cut-pursuit in windows system,visual studio2015 compiler.But it breaks down for the blowed errors in terms of the boost::graph libraries :
[image]
this bug point to codes in API.h：
[image]
I am not sure it is the windows system or the version of boost cause this problem.Can anybody try to do the same compiling procedure like me?Want some HELP,please.

loicland, 2020-06-09:
what version of gcc are you using?

zhairuoming, 2020-06-09:
> what version of gcc are you using?

I use cl.exe and link.exe in microsoft visual studio2015 and it is easier for me to compile and generate in windows system. Is that viable?or just have to use gcc?

zhairuoming, 2020-06-10:
hi,loicland.I try the cut-pursuit with cmake3.11.0 and GCC-6.3.0.The same error happen again in the boost::graph class ,like this:
`D:/Program Files/anaconda3/Library/include/boost/graph/detail/adjacency_list.hpp:2664:18: note:   no known conversion for argument 1 from 'boost::detail::adj_list_edge_iterator<boost::range_detail::integer_iterator<unsigned int>, boost::detail::out_edge_iter<__gnu_cxx::__normal_iterator<boost::detail::stored_edge_property<unsigned int, CP::EdgeAttribute<float> >*, std::vector<boost::detail::stored_edge_property<unsigned int, CP::EdgeAttribute<float> >, std::allocator<boost::detail::stored_edge_property<unsigned int, CP::EdgeAttribute<float> > > > >, unsigned int, boost::detail::edge_desc_impl<boost::directed_tag, unsigned int>, int>, boost::adjacency_list<boost::vecS, boost::vecS, boost::directedS, CP::VertexAttribute<float>, CP::EdgeAttribute<float>, boost::no_property, boost::listS> >::value_type {aka boost::detail::edge_desc_impl<boost::directed_tag, unsigned int>}' to 'boost::adj_list_edge_property_map<boost::directed_tag, CP::EdgeAttribute<float>, CP::EdgeAttribute<float>&, long unsigned int, CP::EdgeAttribute<float>, boost::edge_bundle_t>::key_type {aka boost::detail::edge_desc_impl<boost::directed_tag, long unsigned int>}'
src\CMakeFiles\cp.dir\build.make:62: recipe for target 'src/CMakeFiles/cp.dir/cutpursuit.cpp.obj' failed
make[2]: *** [src/CMakeFiles/cp.dir/cutpursuit.cpp.obj] Error 1
CMakeFiles\Makefile2:84: recipe for target 'src/CMakeFiles/cp.dir/all' failed
make[1]: *** [src/CMakeFiles/cp.dir/all] Error 2
Makefile:82: recipe for target 'all' failed
make: *** [all] Error 2`
Is my bug comes from the boost libraries?I use the boost graph library—libboost_graph-vc140-mt-x64-1_67.lib.Should I use the other library called boost_graph-vc140-mt-x64-1_67.lib?I am confused.

zhairuoming, 2020-06-11:
thanks , loicland!problem is solve by changing the definition of (line64 in graph.h)
`using EdgeAttributeMap = boost::adj_list_edge_property_map<
boost::directed_tag, EdgeAttribute<T>, EdgeAttribute<T> &
,long unsigned int, CP::EdgeAttribute<T>, boost::edge_bundle_t>;`
According to [this issue](https://github.com/loicland/cut-pursuit/issues/14),I first change the 'long unsigned int' to 'unsigned __int64' and it failed .Then I found the definition in boost/graph/detail/adjacency_list.hpp is just 'unsigned int'.I change it to 'unsigned int' and it works.
I'm also confused about the output library 'libcp.dll.a' ,as it is just 'libcp.dll' in windows.I have to manually delete the '.a' suffix. But thanks a lot,you help me solve the big problem!!!

closed 2021-04-22 by loicland