loicland/superpoint_graph #214: problem with Annotations.txt while running partition.py

zhairuoming, 2020-06-13:
I run thie code as 'python partition/partition.py --dataset s3dis --ROOT_PATH datasets/s3dis --voxel_width 0.03 --reg_strength 0.03'
the dataset filepath is E:\8-testdata\Stanford3dDataset_v1.2_Aligned_Version-004\Area_1\
-----conferenceRoom_1.txt
-----Annotations
--------beam_1.txt
--------board_1.txt
--------......
the bug shows that 
`=================
   Area_1/
=================
1 / 1---> Annotations
    creating the feature file...
Traceback (most recent call last):
  File "partition/partition.py", line 122, in <module>
    xyz, rgb, labels, objects = read_s3dis_format(data_file)
  File "/home/ubuntu/MYFILE/superpoint_graph/partition/provider.py", line 189, in read_s3dis_format
    room_ver = pd.read_csv(raw_path, sep=' ', header=None).values
  File "/home/ubuntu/anaconda3/lib/python3.6/site-packages/pandas/io/parsers.py", line 678, in parser_f
    return _read(filepath_or_buffer, kwds)
  File "/home/ubuntu/anaconda3/lib/python3.6/site-packages/pandas/io/parsers.py", line 440, in _read
    parser = TextFileReader(filepath_or_buffer, **kwds)
  File "/home/ubuntu/anaconda3/lib/python3.6/site-packages/pandas/io/parsers.py", line 787, in __init__
    self._make_engine(self.engine)
  File "/home/ubuntu/anaconda3/lib/python3.6/site-packages/pandas/io/parsers.py", line 1014, in _make_engine
    self._engine = CParserWrapper(self.f, **self.options)
  File "/home/ubuntu/anaconda3/lib/python3.6/site-packages/pandas/io/parsers.py", line 1708, in __init__
    self._reader = parsers.TextReader(src, **kwds)
  File "pandas/_libs/parsers.pyx", line 384, in pandas._libs.parsers.TextReader.__cinit__
  File "pandas/_libs/parsers.pyx", line 695, in pandas._libs.parsers.TextReader._setup_parser_source
FileNotFoundError: File b'datasets/s3dis/data/Area_1/Annotations/Annotations.txt' does not exist`
I don't know why it mentioned ' Annotations/Annotations.txt',because the Annotations filefolder do not contain Annotations.txt!

Anyone knows about it?

closed 2020-06-16 by zhairuoming

loicland, 2020-06-16:
I think there must be an error with your pathing or during the download of S3DIS. Did you figure it out?

zhairuoming, 2020-06-16:
> I think there must be an error with your pathing or during the download of S3DIS. Did you figure it out?

yes,sorry.My mistake.By the way,I only add the Area_1 filefolder to the path.The bug shows that 
`No such file or directory: 'datasets/s3dis/superpoint_graphs/Area_2/'` 
 I saw in your code you create all the areas folder
`if args.dataset == 's3dis':
    folders = ["Area_1/", "Area_2/", "Area_3/", "Area_4/", "Area_5/", "Area_6/"]
    n_labels = 13`
So may I have to add all the six Areas to the path to process the spg?
I just want to process one Area like Area_1/conferenceRoom_1.

loicland, 2020-06-16:
If you are only interested in the SPG, you could probably just change the values of `folders` to `["Area_1/"`]`. Let me know if it causes another bug.